GreyGnome/EnableInterrupt #1: struct functionPointerPortB

arjunmehta94, 2015-04-10:
Hello, I get the following error when I try to compile the code. Can you please provide some guidance on this?

In file included from Interrupt.ino:1:
/Users/arjunmehta94/Documents/Arduino/libraries/EnableInterrupt-master/EnableInterrupt.h: In function 'void __vector_3()':
/Users/arjunmehta94/Documents/Arduino/libraries/EnableInterrupt-master/EnableInterrupt.h:976: error: 'struct functionPointersPortB' has no member named 'pinSix'
/Users/arjunmehta94/Documents/Arduino/libraries/EnableInterrupt-master/EnableInterrupt.h:977: error: 'struct functionPointersPortB' has no member named 'pinSeven'

GreyGnome, 2015-04-10:
Sorry about that. Which Arduino are you using?

I will take a look at it.

arjunmehta94, 2015-04-10:
I am using the Uno.

GreyGnome, 2015-04-12:
I added some #ifndef so the problem pins are not attempted to be compiled.
Odd that I didn't see that problem in my testing.

Could you grab the EnableInterrupt.h file from the Github site and give it
a test?

GreyGnome, 2015-05-14:
30 days; autoclose.

closed 2015-05-14 by GreyGnome