GreyGnome/EnableInterrupt #2: Pin port number in called routine

rflowers, 2015-04-15:
Hello,

I'm using your excellent  library to implement RC Pwm timing on an arduino uno. The interrupt routine I use is identical between pins, but needs to store separate data for separate pins. is there anyway for my interrupt routine to easily access which pin generated the interrupt request so I can use the same code for both pin interrupts with an array? 

thanks very much!

GreyGnome, 2015-04-16:
I would create different (small) frontend interrupt functions for different
pins, then each of those functions calls a common function, perhaps with
the pin number as an argument. Thus, if pin 8 gets assigned
"myInterruptFunction8()", then in that function you would call
"commonInterruptFunction(8);" .

Funny, I had a variable "arduinoPin" in the old PinChangeInt library but it
was broken for years :-) . I was thinking that it was useless . But now
that I created a new library, you need this feature.  :-)

How wide are your PWM pulses? Calling the second function may add
microseconds to the code and that may be fine. You could inline the common
function, but that uses memory. A third option would be for me to include
the functionality, but that slows down the library for everyone.

Every choice has a tradeoff.

rflowers, 2015-04-18:
The pulses are measured as 1000 - 2000 microseconds on pins 2-7. I'm currently inlining the timing routines but would love to shrink the code. I hoped you were implementing the actual pin interrupt handler by deciding which pin had actually activated, and then using this as an offset to a lookup table of attached routines, so it would be simple to make this offset available to external code, but if you've implemented it differently, no matter.

I think some kind of function to know which pin had been activated would be useful for testing purposes and demonstrating how to use the library. I appreciate that you wouldn't want to embed this in your base code, but as a library function it could help.

thanks

GreyGnome, 2015-05-14:
I'm still pondering this one. Right now I'm working on Sanguino (aka ATmega644/1284) support. But a pin number should be relatively easy to implement, although I fear the #define's will make the code ugly. We'll see...

rflowers, 2015-05-14:
thanks that would be great. of mention the library clashes with spi.h, soft serial.h and servo.h, some kind of vector is used by both. I'm not sure why but I suspect it is due to them hogging interrupt routines. Your code helped fly a UAV on Monday though, so pretty cool thanks!

GreyGnome, 2015-07-25:
Ok, I have been working on this and I note there are a couple of ways I can go:

First, I can set the pin number in a variable, call it "arduinoInterruptedPin". Then a common interrupt routine gets called and can query that variable to discover which pin called it.

It would look like this:

    #define EI_ARDUINO_INTERRUPTED_PIN
    void interruptFunctionA() {
      switch (arduinoInterruptedPin) {
        case 10:
          (do something)
          break;
        case 11:
          (do something)
          break;
      }
    }

Second, I can increment a variable associated with each pin. This capacity already exists and is used by the high speed mode to make it very fast. I just could extend the normal mode and increment a variable named by the programmer. It would look like this:

    #define INTERRUPT_FLAG_PIN10 pin10flag
    #define INTERRUPT_FLAG_PIN11 pin11flag

    void interruptFunctionB() {
      if (pin10flag) {
        (do something)
        pin10flag=0;
      }
      if (pin11flag) {
        (do something)
        pin11flag=0;
      }
    }

The first technique looks more eloquent in C, though it complicates the library code. But the assembler is actually a little more efficient: it loads the arduinoInterruptedPin once from memory, then uses a cpi instruction (1 cpu cycle), as long as the case argument is less than 256. For the latter method, each flag must be loaded from memory for each if statement so there will always be at least two lds statements.

Therefore, I will provide the "arduinoInterruptedPin" variable.

GreyGnome, 2015-08-08:
Version 0.8.0 is now out, which has the interrupted pin feature. See http://forum.arduino.cc/index.php?topic=313323.new#new

closed 2015-08-08 by GreyGnome