GreyGnome/EnableInterrupt #3: conflict please help.

shirish47, 2015-05-07:
I am using your library which performs great. But I wanted to use it with nrf24. For nrf Radio modules I am using radiohead's library and they both seem to have some conflict. I checked that it has to do with the driver RH_RF24.h because commenting it causes no error. 
error is .
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
C:\Users\ASUS-PC\AppData\Local\Temp\build6046962854557047475.tmp/core.a(WInterrupts.c.o): In function `__vector_1':
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/WInterrupts.c:309: multiple definition of `__vector_1'
powerDownWakeExternalInterrupt_MY.cpp.o:C:\Program Files (x86)\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:948: first defined here
C:\Users\ASUS-PC\AppData\Local\Temp\build6046962854557047475.tmp/core.a(WInterrupts.c.o): In function `attachInterrupt':
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/WInterrupts.c:39: multiple definition of `__vector_2'
powerDownWakeExternalInterrupt_MY.cpp.o:C:\Program Files (x86)\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:473: first defined here
collect2.exe: error: ld returned 1 exit status
Error compiling.
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
I would also try to contact radio head's author.
Any Idea How I resolve this. 
I am attaching interrupts on  2,3,4,5,6 pin on Pro mini. Also using a LowPower.h but thats not causing this I have check. 
Thanks in advance.

GreyGnome, 2015-05-07:
Pins 2 and 3 are taken by the RadioHead library to use for its interrupt
processing... I haven't dug into it but I suppose that's where you're
supposed to connect some of the pins? Anyway I don't think you can use them
in your sketch, if that was your plan.

Regardless, yes the libraries are colliding. At this point the
EnableInterrupt library defines those ISRs, and the RadioHead library uses
attachInterrupt() which defines those ISR's.. If you want, you can try to
comment out the External Interrupt ISRs (those are the one's used by pins 2
and 3 and defined by RadioHead). Simply look for the line in the
EnableInterrupt.h file that says:

ISR(INT0_vect) {
Above it, put simply
/*
and then search down to the end of the ISR(INT1_vect) subroutine. It will
be the first closing brace } that is after:

ISR(INT1_vect) {
Below that closing brace, put

*/

That should comment out the two conflicting ISR's. You can use other pins
instead of 2 and 3, if you need to, with Pin Change interrupts.

I should be able to add a #define that would allow you to exclude
conflicting ISR's like this, but it's not a high priority right now.

Let me know how it goes.

shirish47, 2015-05-08:
Well it compile Now, thanks. :) Some how people at arduino should do something so that people can test their libraries compatibility with other. That would be helpful to those who are using multiple of them. Most of these people don't understand hardware side.

shirish47, 2015-05-14:
will this library only work with push buttons or can it work with signals coming from some chip or something.

GreyGnome, 2015-05-14:
The library, and the arduino, can't tell the difference between a
pushbutton and a signal generated in some other fashion. They are all just
signals.

Electrically, however, you may need to be aware of things like the forward
voltage drop across an LED so you want to make sure your "0" level is low
enough and your "1" level is high enough for the atmega chip to properly
recognize it.

In general, if you turn on the input pullup resistor on the pin and you
have a digital chip talking to the arduino, you will be fine.

GreyGnome, 2015-06-18:
I'm going to close this issue. In the future I will have support for commenting out unneeded chunks easily...

closed 2015-06-18 by GreyGnome

shirish47, 2015-06-19:
Thanks for the help. :)

It was great working with this lib.