GreyGnome/EnableInterrupt #4: Conflict with SoftSerial

K6TD, 2015-05-19:
Including both Softserial and EnableInterrupt produces a compile conflict.  Both libraries appear to be going after the same interrupt vector.

K6TD, 2015-05-19:
Arduino: 1.6.4 (Mac OS X), Board: "Arduino Leonardo"

...

L/var/folders/xm/zr3tcs311qb068gnbjq0_73m0000gn/T/build3350661289325773792.tmp -lm 
/var/folders/xm/zr3tcs311qb068gnbjq0_73m0000gn/T/build3350661289325773792.tmp/SoftwareSerial/SoftwareSerial.cpp.o: In function `__vector_9':
/Applications/Arduino_164.app/Contents/Java/hardware/arduino/avr/libraries/SoftwareSerial/SoftwareSerial.cpp:227: multiple definition of `__vector_9'
/var/folders/xm/zr3tcs311qb068gnbjq0_73m0000gn/T/build3350661289325773792.tmp/GPS_1PPS.cpp.o:/Users/kevin/working/Ham Radio/IBPV2/IBPV2_Rev2/libraries/EnableInterrupt/EnableInterrupt.h:1087: first defined here
collect2: error: ld returned 1 exit status
Error compiling.

GreyGnome, 2015-05-19:
Yes, there is no way to build a library that may provision an interrupt vector and look for other definitions of an interrupt vector... vector_9, from /usr/avr/include/avr/iom32u4.h, shows that it is the ISR for PCINT0, which is the Pin Change Interrupt ISR. This means that you can only use External Interrupt pins with the Leonardo. You must comment out lines 1086-1124 in EnableInterrupt.h (essentially, the entire ISR that begins with "ISR(PORTB_VECT) {" and ends at the closing brace "}" ) and ensure that you are not using any pin change interrupt pins. See https://github.com/GreyGnome/EnableInterrupt/wiki/Usage#Arduino_Leonardo . The only pins available to you from this library are 0-3 and 7.

I actually suggest you forgo using this library and just attachInterrupt() on your Leonardo.

Technically, it may be possible for the two libraries to interact but they both gobble up the interrupt system on the Arduino. 

BTW, have you considered http://www.pjrc.com/teensy/td_libs_AltSoftSerial.html ? Besides requiring the signals on certain pins, it appears to be compatible with this library (it doesn't use Pin Change Interrupts).

closed 2015-05-19 by GreyGnome

K6TD, 2015-05-19:
Thanks for the pointers and understanding of the conflict issue.  No clear way to share the interrupt vector.