GreyGnome/EnableInterrupt #5: Compatibility with Arduino IDE library manager

jan-van, 2015-05-30:
Nice Lib. I had to move around the folders and files a little before my scetch compiled.
The current version of the Arduino IDE (1.6.4) has a standardised method to include external libraries.
By default it places the files from a ZIP file into a directory under the MyDocuments/Arduino folder.

When included using the libarary manager, it expects the header file to be included in the top folder of the libary file, and that header file is then inserted in the source of the sketch. Thus it requires the "EnableInterrupt.h" to be moved to the top folder of the ZIP file. Only the file to be included in the sketch is needed since all .h files found in the top folder are inserted in the sketch source. The description is currently not visible in the list within the IDE.

A manual on how to organise external libraries is documented on http://www.arduino.cc/en/Guide/Libraries

GreyGnome, 2015-06-01:
Thanks, I'll look into it.

GreyGnome, 2015-06-18:
I'm working on this now.

GreyGnome, 2015-06-23:
The library is in the Library Manager. It looks good to me... let me know if you have any further issues.

Thanks for the suggestion.

closed 2015-06-23 by GreyGnome