GreyGnome/EnableInterrupt #6: How I can use enableInterrupt in class

k-samsonov, 2015-06-17:
// Test.h

#ifndef _TEST_h
#define _TEST_h

#if defined(ARDUINO) && ARDUINO >= 100
	#include "arduino.h"
#else
	#include "WProgram.h"
#endif

#define ADC_EXTERNAL_PIN 2

#include <EnableInterrupt.h>

class TestClass
{
 protected:
  interruptFunctionType pInt;

 public:
  TestClass();
	void init();
  void start();
  void stop();
  static void _interruptFunction();
};

#endif

// Test.cpp

#include "Test.h"

void TestClass::init()
{

  pInt = static_cast<interruptFunctionType>( TestClass::_interruptFunction);
}

  void TestClass::start()
  {
    enableInterrupt (ADC_EXTERNAL_PIN,  pInt  , FALLING);    
  }
  void TestClass::stop()
  {
    disableInterrupt (ADC_EXTERNAL_PIN);    
  }

this code not compiled

NicoHood, 2015-06-17:
You have to define such a libcall before the include:
https://github.com/GreyGnome/EnableInterrupt/blob/master/EnableInterrupt.h#L105

And you have to include the library of course too. In the header and in the .ino.

You might also want to try this:
https://github.com/NicoHood/PinChangeInterrupt

GreyGnome, 2015-06-18:
Essentially you want to mix C and C++. The answer lurks somewhere in here: https://isocpp.org/wiki/faq/mixing-c-and-cpp

On my todo list is to create some sample code that folks can use as a template.

If you decide to try the PinChangeInterrupt library, realize that it has a different focus than the EnableInterrupt library. It only works on pin change interrupts on ATmega processors, and you have to translate between the pin you want and a "pcint". For the EnableInterrupt library you attach an interrupt to an Arduino pin that supports interrupts (sorry- I'm no magician. Can't attach an interrupt to a pin that doesn't support them :-)  ). The EnableInterrupt library's intention is to keep you at the Arduino level and not worry about the underlying mechanism of the processor.

There are other differences: PinChangeInterrupt may save a couple of dozen bytes of RAM on the larger processors (ie, 328 and above) over the EnableInterrupt library (another on my list of todo's is to eliminate this waste for those who care).  PinChangeInterrupt works on ATtiny chips; EnableInterrupt is more Arduino-centric. There are optimization differences as well.

Speed-wise they would be about the same; let's put it this way: If you are using digitalWrite() then the speed of either of these interrupt libraries is probably more than adequate for you.

GreyGnome, 2015-06-23:
I have created a new release and it now includes an example, OOSimple.ino. Check it out.

Briefly, you are mixing C and C++, and the EnableInterrupt library is C code. So set up the interrupts in C, and have it call a C++ method in an object. You can then perform other logic in the C++ code, but largely the interrupt library is managed mostly in C.

It looks like it's pretty easy to enable the interrupt from the C++ object. See http://stackoverflow.com/questions/16850992/call-a-c-function-from-c-code, or, of course, Google. I'll leave the details of that as an exercise for the reader :-) .

GreyGnome, 2015-06-23:
The example is written and tested. This issue is closed.

closed 2015-06-23 by GreyGnome