GreyGnome/EnableInterrupt #7: EnableInterrupt with SoftwreSerial

hariprasadiit, 2015-06-24:
cant use both as both are defining same vectors.is there a way to use them both by modifying?

GreyGnome, 2015-06-28:
I am at this moment working on putting flags in the EnableInterrupt library so you could do, for example:

'''
#define EI_NOTPORTC
'''
and the library would leave the ISR for port C alone entirely. Look for it soon.

GreyGnome, 2015-07-07:
Version 0.7.0 has been released which should cover this issue. See https://github.com/GreyGnome/EnableInterrupt/wiki/SaveMemory. You'll need to determine which interrupts the other library defines.

BTW, I have to say that the SoftwareSerial is, on first glance, terrible. When an interrupt is called on a receive it goes into a routine that receives all 8 bits of a serial transmission. Interrupts are off during this time period. This period is measured in milliseconds, which is terrible for an ISR. If you expect the CPU to service other switches or what have you... well, I wouldn't.

AltSoftSerial at https://github.com/PaulStoffregen/AltSoftSerial seems much better.

And that's all I have to say about that. Caveat Programmer.

hariprasadiit, 2015-07-07:
thanks for the update. I'll try to use AltSoftSerial.thanks for great and simple library.

closed 2015-07-07 by hariprasadiit