GreyGnome/EnableInterrupt #8: Interrupt Numbers with Mega2560 correct?!

Apollon77, 2015-06-26:
In the pint-to-INT-List I see different things then in the official infos.

Official:
External Interrupts: 2 (interrupt 0), 3 (interrupt 1), 18 (interrupt 5), 19 (interrupt 4), 20 (interrupt 3), and 21 (interrupt 2). These pins can be configured to trigger an interrupt on a low value, a rising or falling edge, or a change in value. See the attachInterrupt() function for details. 

this differs from your infos ... will it work on mega or not?

GreyGnome, 2015-06-28:
Yes, it will work. I'm not sure why attachInterrupt() was designed with "int.0", "int.1", etc. This is part of the reason why I found interrupts so confusing, and wanted to create a new library- it keeps everything consistent (and out of sync with attachInterrupt(), btw).

Let's look at "int.2" from the attachInterrupt() function for an example. int.2 is on Arduino pin 21 of the Mega2560. Arduino Pin 21 is ATmega2560 pin 43, or PD0/INT0. But if it's INT0, why int.2? The crux of the problem.

The mode of ATmega2560 INT0 is set by the first 2 bits of EICRA (ISC00 and 01). You also set bit 0 of EIMSK to enable the interrupt.

If you look at the attachInterrupt() function code, the interrupt number is used in a case statement to set the proper EIMSK and EICRA bits. For int.2, this code looks like:
```
    case 2:
      EICRA = (EICRA & ~((1 << ISC00) | (1 << ISC01))) | (mode << ISC00);
      EIMSK |= (1 << INT0);
      break;
```

...The code is setting ISC00 and ISC01 to mode. It is also putting a 1 in bit 0 of EIMSK. This corresponds to INT0 of the ATmega2560.

Again, I don't understand attachInterrupt()'s "int.X" numbering scheme. When I write code, I try not to always concern myself with the arcane details of the chip... I simply want to add a device to a pin and perform some action. That's why EnableInterrupt() uses pin numbers; to allow the programmer to assign interrupts in the problem domain (that is, the Arduino pin number) rather than the CPU domain (which requires you to always keep a table nearby to convert that information to an Arduino pin number).

In short, EnableInterrupt will work on a Mega. The rest is academic.

GreyGnome, 2015-06-28:
EnableInterrupt() works. attachInterrupt() is odd. Issue closed.

closed 2015-06-28 by GreyGnome

Apollon77, 2015-06-28:
Thank you for the infos. great!!