GreyGnome/EnableInterrupt #9: Passing some form of state to interrupt subroutines

DanielOgorchock, 2015-06-27:
Do you think it's feasible to change the callback function type to a void function that has a void pointer parameter? This would allow for passing state to the callback function, eliminating the need to declare multiple functions for each instance of a particular object (as your ooSimple example would require). I know that storing the void pointers may roughly double the library's memory usage and may complicate the #defines. Regardless, I believe this would be really useful.

DanielOgorchock, 2015-06-27:
Also, it would eliminate the issue raised in #2

GreyGnome, 2015-06-28:
Let me think on this one for a little while. The existing callback function form must stay in place, because the Arduino Style guide requires that I keep beginners in mind for this library. Specifically, it states (all on its own line):

Avoid pointers

So that's a big deal. Regardless, it would probably be possible to add a #define flag that would turn on this functionality, for the more technically savvy that needed this.

GreyGnome, 2015-07-09:
Ok, I'm ready to look at the "get-state-information-from-ISR's" issue.

The way I see it is this:

If I create the callback so that it allows for a pointer parameter, I think it will look something like:

    void (*interruptFunction)(void *genericpointer);

...or similar. I'm assuming that the end user would define the genericpointer. Thus, for every interrupt pin added to the system we would need a call like this:

    enableInterrupt(uint8_t pinNumber, void (*userFunction)(void *genericpointer), void (&genericthingy), uint8_t mode);

...so that inside the ISR, the genericthingy could be assigned to the genericpointer so that when the userFunction is called, the proper genericthingy would be given as an argument to the userFunction.

I don't see that it's any different than allowing the ISR to assign a value to a variable that each user function could query, to find out which pin triggered it. The user's ISR could query the pin and find out if it went LOW or HIGH, if necessary.

Pin number + level change is pretty much all that the library could ever provide; that is the definition of the interrupt (Which pin? What happened?). Everything else is up to the programmer's ISR that the library would call. So the programmer could assign a single function to all interrupted pins and have a switch statement that performs x, y, or z depending on pin.

But I feel like I'm missing something. Could you let me know what you had in mind?

At this point I'm in favor of making available some kind of state information. I would be inclined to only provide the pin number, but I could do something else if it's better. I just don't see it at the moment.

DanielOgorchock, 2015-07-10:
I understand what you're getting at, and I think you're probably right. A simple pin number variable should allow the user to do whatever they want in his or her function without overcomplicating the library.

The generic pointer approach would have some advantages. For example, users could create a class that passes a pointer of itself in its constructor to the enableInterrupt function. This would allow for an arbitrary number of class instances waiting for interrupts on an arbitrary number of pins, not requiring a switch statement or iterating through an array of said objects, doing integer comparisons on a pin variable to find the proper one. (For example, I've been working on a radio controlled flight controller, and being able to add a new radio channel by just writing RadioChannel channel4(9); would be convenient)

That said, I guess in the realm of arduino the programmer typically knows the exact number of objects being created since there's such limited RAM. I think the pin variable's simplicity probably outweighs the minor advantages of the void pointer approach.

GreyGnome, 2015-07-14:
You bring up some nice functionality, and I've been thinking about this some more but I think that adding C++ features to the library is out of scope. I had written ooPinChangeInt (an object-oriented version of the PinChangeInt library) and although it was nice, it did add significant latency to the interrupt. I'm going for something else here, which is to be closer to the iron so to speak, and thus keep it small and closer to pure assembly speed.

So, to finalize: I'll add the pin number variable and close out this issue.

GreyGnome, 2015-08-08:
Version 0.8.0 is now out, which has the interrupted pin feature. See http://forum.arduino.cc/index.php?topic=313323.new#new

closed 2015-08-08 by GreyGnome