GreyGnome/EnableInterrupt #10: in v 0.7, cannot disable port D for arduino UNO

hwestbrook, 2015-07-15:
my setup: 
```
#define EI_NOTPORTB
// #define EI_NOTPORTC 
#define EI_NOTPORTD
#define EI_NOTEXTERNAL
#include <EnableInterrupt.h>
```
errors: 
```
/Users/hwestbrook/Documents/Arduino/libraries/EnableInterrupt/EnableInterrupt.h: In function 'void __vector_5()':
/Users/hwestbrook/Documents/Arduino/libraries/EnableInterrupt/EnableInterrupt.h:1192:19: error: 'portSnapshotD' was not declared in this scope
   changedPins   = portSnapshotD ^ current;
                   ^
/Users/hwestbrook/Documents/Arduino/libraries/EnableInterrupt/EnableInterrupt.h:1193:19: error: 'risingPinsPORTD' was not declared in this scope
   tmp           = risingPinsPORTD & current;
                   ^
/Users/hwestbrook/Documents/Arduino/libraries/EnableInterrupt/EnableInterrupt.h:1194:19: error: 'fallingPinsPORTD' was not declared in this scope
   interruptMask = fallingPinsPORTD & ~current;
                   ^
/Users/hwestbrook/Documents/Arduino/libraries/EnableInterrupt/EnableInterrupt.h:1205:31: error: 'portDFunctions' was not declared in this scope
   if (interruptMask & _BV(0)) portDFunctions.pinZero();
                               ^
/Users/hwestbrook/Documents/Arduino/libraries/EnableInterrupt/EnableInterrupt.h:1206:31: error: 'portDFunctions' was not declared in this scope
   if (interruptMask & _BV(1)) portDFunctions.pinOne();
                               ^
/Users/hwestbrook/Documents/Arduino/libraries/EnableInterrupt/EnableInterrupt.h:1207:31: error: 'portDFunctions' was not declared in this scope
   if (interruptMask & _BV(2)) portDFunctions.pinTwo();
                               ^
/Users/hwestbrook/Documents/Arduino/libraries/EnableInterrupt/EnableInterrupt.h:1208:31: error: 'portDFunctions' was not declared in this scope
   if (interruptMask & _BV(3)) portDFunctions.pinThree();
                               ^
/Users/hwestbrook/Documents/Arduino/libraries/EnableInterrupt/EnableInterrupt.h:1209:31: error: 'portDFunctions' was not declared in this scope
   if (interruptMask & _BV(4)) portDFunctions.pinFour();
                               ^
/Users/hwestbrook/Documents/Arduino/libraries/EnableInterrupt/EnableInterrupt.h:1210:31: error: 'portDFunctions' was not declared in this scope
   if (interruptMask & _BV(5)) portDFunctions.pinFive();
                               ^
/Users/hwestbrook/Documents/Arduino/libraries/EnableInterrupt/EnableInterrupt.h:1211:31: error: 'portDFunctions' was not declared in this scope
   if (interruptMask & _BV(6)) portDFunctions.pinSix();
                               ^
/Users/hwestbrook/Documents/Arduino/libraries/EnableInterrupt/EnableInterrupt.h:1212:31: error: 'portDFunctions' was not declared in this scope
   if (interruptMask & _BV(7)) portDFunctions.pinSeven();
                               ^
```

It compiles just fine if port c is disabled too

GreyGnome, 2015-07-16:
Whoops... typo by me. I believe the problem will be fixed if you find this line:

    #ifndef EI_NOTPORTC
    ISR(PORTD_VECT) {

and replace it so:

    #ifndef EI_NOTPORTD
    ISR(PORTD_VECT) {

I am working on the "report pin" functionality and I will roll in this fix with the next release.

GreyGnome, 2015-08-08:
Version 0.8.0 is out now and it includes this fix. Let me know if it's still broken. I'll leave this issue open for a week or two...

GreyGnome, 2015-08-17:
Closed due to inactivity. Presumed fixed.

closed 2015-08-17 by GreyGnome