GreyGnome/EnableInterrupt #11: function name conflict with SD card library

zcattacz, 2015-07-16:
```
In file included from recTest_Calculate.ino:16:0:
~\Documents\Arduino\libraries\EnableInterrupt-master/EnableInterrupt.h: In function 'void SerialPrint_P(const char*)':
~\Documents\Arduino\libraries\EnableInterrupt-master/EnableInterrupt.h:115:6: error: redefinition of 'void SerialPrint_P(const char*)'
 void SerialPrint_P(const char *str) {
      ^
In file included from C:\Program Files\Tools\arduino-1.7.3\libraries\SD\src/SD.h:21:0,
                 from recTest_Calculate.ino:5:
C:\Program Files\Tools\arduino-1.7.3\libraries\SD\src/utility/SdFatUtil.h:61:22: error: 'void SerialPrint_P(const char*)' previously defined here
 static NOINLINE void SerialPrint_P(PGM_P str) {
                      ^
Error compiling.

```

GreyGnome, 2015-07-16:
Ooops. This will be fixed soon. In the meantime, you can simply change the name in the two places it appears in the .h file: lines 114 and 115.

GreyGnome, 2015-08-08:
Version 0.8.0 is now out, which has the interrupted pin feature and this fix.. See http://forum.arduino.cc/index.php?topic=313323.new#new

closed 2015-08-08 by GreyGnome