GreyGnome/EnableInterrupt #12: Support for ATtiny45/85 and ATtiny44/84

SukkoPera, 2015-07-17:
Is there any chance to get these chips supported in the library?

They all have a single external interrupt pin, pin change interrupts on all pins and are supported in the Arduino IDE through David A. Mellis's cores available at https://github.com/damellis/attiny.

See http://fc04.deviantart.net/fs70/f/2013/038/3/7/attiny_web_by_pighixxx-d5u4aur.png for reference.

GreyGnome, 2015-07-21:
I would imagine that you would not want to use this library to enable interrupts on those smaller chips, as (for example) the ISR code will use memory and the user ISR pointer arrays could waste 14 bytes which is 5% of available RAM on a 256-byte chip.

Are programmers more interested in the ease of use/cross-platform functionality, that the tradeoff is worth it on the littler chips?

SukkoPera, 2015-07-21:
In my opinion programmers might prefer portability on different chips or some more free memory, depending on the context.

In my case I have a heavily-#ifdef'd program, a stripped-down version of which can easily be built for Attiny85's. Two interrupts are critical for its working: one is needed to detect a quick clock signal and must be bound to a hardware interrupt, while the other one works fine even through a pin change interrupt. Memory is not a concern, as the whole program is using some 30 bytes, but being able to use this library on all platforms would be great to reduce the #ifdef burden a bit, although I understand that this is something not everybody might need/desire.

It's up to you :).

GreyGnome, 2015-07-23:
Ok thanks. Yeah I'm in #ifdef hell right now with this thing. I am looking at simplifying the situation, as another platform will make my particular hell all that much more uncomfortable :-) . Anyway, I'll put this in the queue.

SukkoPera, 2015-07-23:
Thanks a lot! If you think I might help, just ask!

GreyGnome, 2015-08-08:
Any help is appreciated. If you want to do the coding, by all means... :+1:

GreyGnome, 2015-08-17:
I'm working on this right now. I hope to have a release in a week.

GreyGnome, 2015-09-30:
...Did I say week? I meant, "A month and a half". It's almost ready for release, I think.

GreyGnome, 2015-10-08:
It should be up now in version 0.9.0

SukkoPera, 2015-10-12:
Well, and did I say I would help? I meant, "Maybe with the testing" :). Hopefully I will be able to do something soon. Thanks, in the meantime!

GreyGnome, 2015-10-16:
Ok good luck. I now have an ATtiny85 and have messed around with ISP on an Arduino Pro Mini. So doing the ATtiny should not be too hard. Anyway, let me know if you notice anything. In the meantime I will close this issue. Enjoy!

closed 2015-10-16 by GreyGnome