GreyGnome/EnableInterrupt #13: This library does not coexist with other code using interrupts

TomWS1, 2015-07-19:
I tried to use this library to add pinChange interrupt capability to an existing sketch.  When I simply added #include <EnableInterrupt.h> into my main sketch, the build failed with multiple duplicate references.  
After exploring a bit I discovered that all I had to do was use the 'Simple' example, #include a reference one of my other libraries (which supports hardware with its own interrupt handler) and that build failed.   I then created a very simple extension to 'Simple' example in which there was a secondary file with its own interrupt handler and was not surprised that that failed as well.  (I'd include the source files here but don't see how to do it)

Also, the EnableInterrupt.h file is peppered with references to actual Arduino boards, not to the underlying processor.   Consequently any board that is NOT included in the set will not be built properly, these include new Arduino designs or designs using AVR processors that are not 'official' Arduino designs.

GreyGnome, 2015-07-21:
Yes, I am working on adding compiler directives to exclude interrupt handlers that may conflict with handlers in other libraries. The problem is that other libraries may consume ISRs that they don't need, so you may have to edit their files. Note that the AltSoftSerial library avoids some of those issues that are caused by the SoftwareSerial library (and it is superior, IMHO).

Look for a new version in the next week or two. You can download the EnableInterrupt.h file today which will have the directives in it already.

To add files you could simply include a link to Github Gist or pastebin.

The EnableInterrupt.h is peppered with references, but that's only to simplify the #ifdef's in the file. For example, I have supported the non-Arduino ATmega1284 by doing this:

    #elif defined __AVR_ATmega1284P__ || defined __AVR_ATmega1284__ || defined(__AVR_ATmega644P__) || defined(__AVR_ATmega644__)
    #define MIGHTY1284
Similarly, I look for the processor definitions and then define an Arduino name so that further ifdef's don't require that long ugly string.

TomWS1, 2015-07-21:
I'll try to attach the zip file to this reply.

I misunderstood your Arduino references, I thought they were names of 
boards, although they sure do look like that.  Preceding names with a 
double underscore or some prefix unique to your file would avoid future 
confusion.

Re the conflicts, they exist because the executable code is in the .h 
file.  If any other module includes the same .h file, there will be 
conflicts.  It would be better, IMO, to have a distinct class so that 
there is only one instance and all modules use the same instance.

Tom

GreyGnome, 2015-07-21:
I saw your code in the forum at LowPowerLab. (The moteino looks cool, btw. And it's cheap :-) ).

PinChangeInt used a class and it was much slower than this library. That's dangerous to a Pin Change interrupt, because of the latency between the interrupt and actually measuring the pins to see who changed. And it's also miserable for anyone who needs high speed in an interrupt routine, which should be almost everyone.

When the PinChangeInt was in a .h and .cpp file, the Arduino tried to compile it twice. I have a technique that you can use- #define LIBCALL_ENABLEINTERRUPT - that if you put it ahead of your #include in all files but the sketch, then you should be able to use the library without conflicts.

If you're interested, let me know if you try it and it works or not.

TomWS1, 2015-07-21:
Yes, I like Moteino, it's a lot of function in a small quality package 
with fantastic range yet long battery life (under the right conditions).

When you say 'slow', that it usually open to interpretation.  What's 
your definition?  Mine is that I can wake up my device within a couple 
of milliseconds on a pinChange...  I doubt it's that slow. I'm not 
trying to count 100K RPM for an optical pickup to give a ridiculously 
FAST example.

Finally, if I understand your comment re LIBCALL..., if I put this 
#define in my SimplePlus secondary .cpp file I should be able to build 
and execute the additional pinChange interrupt?

Finally finally ;-), thanks for getting back to me.

Tom

GreyGnome, 2015-07-23:
My definition of slow is the ATmega interrupt system. :-) Compared to the ARM Cortex which has a proper hardware-register-save, the ATmega could use 4 microseconds just to enter and another 4 to exit the ISR. And if you've created a library like this, then there would be another function call that the programmer has set up (ie in enableInterrupt() ), that the library will need to call. That's another couple of micros most likely. So you may be up to 10-12 micros before you've even done anything.

The PinChangeInt object-oriented library's ISR took 39 microseconds to perform a simple port bit change, if I recall correctly. That means that the highest frequency signal it could respond to was 1/39micros, or 25khz. If you have multiple interrupts on the same chip things can get really bogged down really quickly.

Regarding your question: Yes put the #define ahead of any #include of this library- EXCEPT for inside your main sketch, so the compiler can actually compile the code- and it should work. If it does not, let me know- that's what I did on the PinChangeInt library so if I screwed up it should be a fairly straightforward fix.

TomWS1, 2015-07-23:
I tried using the #define LIBCALL.. addition to the external files in my Simple case and, if I use board=Moteino (ATmega328p) the Simple case built without a problem.

If I use board=MoteinoMega (ATmega1284p) then the Simple case failed to build due to 'unknown Board' (which probably lead me to believe that the issue was with Board definitions).  It turned out that there are two typos in the EnableInterrupt.h file where MIGHTY1284 was mistyped as MIGHT1284 (lines 699 & 833).  Correcting these allowed the Simple case to build.

However, when I added #include <RFM69.h> (Moteino radio library) to the Simple.ino file, the build failed with multiple definitions for __vector....  I expected this, but when I added #define LIBCALL... & #include <EnableInterrupt.h> to the RFM69.cpp file, the build still failed with same error.  I tried a few permutations of ordering these two statements with respect to #include <RFM69.h> but all of them failed in the same way.  It also didn't matter which board I chose (the only difference was the number of vectors that failed).
If I get some more time to work on this I'll try to isolate the cause, but I'm booked solid at the moment.
Tom

TomWS1, 2015-07-23:
I accidentally pressed the close button, sorry for the mistake .

GreyGnome, 2015-08-08:
Sorry, I just added this to version 0.8.1. Let me know if it's still broken. If so, please send me your sketch so I can verify it. Thanks.

GreyGnome, 2015-08-17:
I will close this by the end of this week if there are no further updates.

TomWS1, 2015-08-17:
The new version of the library seems to enable ATMega1284 ok now, but it 
still generates a conflict if I include the RFM69 library used on the 
Moteino.  I've attached a set of files that will generate the conflict 
with a reference to the RFM69 library in the comment to the #include 
statement.

Tom

GreyGnome, 2015-08-21:
Sorry I don't see the attachment. Maybe you could post it to a gist? (gist.github.com) Or email me. You can reach me at user mschwage on the domain formally know as gmail.com.

GreyGnome, 2015-10-02:
We had a conversation offline about this issue. Tom was able to get his code to compile by taking out some additions to the RFM69.cpp file:
```
//#define LIBCALL_ENABLEINTERRUPT
//#define EI_NOTEXTERNAL
//#include <EnableInterrupt.h>
#include <RFM69.h>
#include <RFM69registers.h>
#include <SPI.h>
```
...he also put in 
```
#define EI_NOTEXTERNAL
#include <EnableInterrupt.h>
#include <RFM69.h>
```
in his sketch. This avoids the two libraries both defining the same ISRs.

closed 2015-10-02 by GreyGnome

martinpineyro, 2016-10-10:
Hi, I know this issue is closed but I am running into this exact problem. 

I am using a Moteino Mega running LMIC library for communication, lowPower for sleeping the MCU and EnableInterrupt to wake the MCU when changes on any of the two sensors occurs. This works perfectly and now I am focusing on saving as much power as I can. 

I read that the RFM69 library has a specific method (sleep()) for putting the radio module into sleep, so I decided to include the RFM69 library just for using that method (none of the communication functionalities of this library will be used), and that was when I came across this issue. Following the instructions provided here I was able to get my code to compile but onChange interrupts wont work (which totally makes sense considering the EI_NOTEXTERNAL definition I added). 

Do you think it is possible to avoid this conflict preserving the external interrupts?

Thanks!

GreyGnome, 2016-10-14:
In order to do what you want to do I think you'll need to hack the RFM69
library to shut off the interrupt sections, if you don't need them. I can't
look at that library right now so that's just a guess.

It should be pretty straightforward- but if and only if those routines are
not needed by that library.