GreyGnome/EnableInterrupt #14: Assigning Class member as interrupt Function.

stickbreaker, 2015-08-01:
GreyGnome,
   I'm in the process of adding CTS/RTS handshaking to the HardwareSerial object.  I had previously used pinInterrupt with arduino 1.0.  With then New 1.6.5 they have totally changed Hardware Serial. The TX buffer is now a member of the class instead of a Global buffer.  
  The complier will not allow (interruptFunctionType)&this->_cts_clear_irq.  It will allow (interruptFunctionType)&HardwareSerial::_cts_clear_irq.  But the instance pointer (this) is not set.  
I am at a loss on how to correctly access these internal buffers from a irq.

any suggestions?
Chuck.

GreyGnome, 2015-08-02:
stickbreaker, with the next release the callback function will be able to access a variable called "arduinoInterruptedPin" which will be the pin number that triggered the interrupt. Will this suffice?

stickbreaker, 2015-08-02:
Yep, I can use that.  
Thanks.

Chuck.

closed 2015-08-02 by stickbreaker

GreyGnome, 2015-08-11:
This feature has been added. See the Usage wiki page for details.