GreyGnome/EnableInterrupt #15: Software Serial

shirish47, 2015-08-13:
Will softwareserial cause problem with this library? 
We are building our hardware and trying to shift UART operations to software serial. But its also necessary for use to use your library. But I want make sure before I can go further. I think SoftwareSerial also use PCINT.

shirish47, 2015-08-13:
I tested this simple thing and I get this..
#include <SoftwareSerial.h>

#include <EnableInterrupt.h>
#include <pindefs_speed.h>
#include <portb_speed.h>
#include <portc_speed.h>
#include <portd_speed.h>
#include <portj_speed.h>
#include <portk_speed.h>

SoftwareSerial mySerial(10, 11);
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

C:\Users\ASUS-PC\AppData\Local\Temp\build5893365334162908593.tmp\SoftwareSerial\SoftwareSerial.cpp.o: In function `__vector_3':
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial/SoftwareSerial.cpp:227: multiple definition of `__vector_3'
C:\Users\ASUS-PC\AppData\Local\Temp\build5893365334162908593.tmp\sketch_aug13a.cpp.o:C:\Program Files (x86)\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:1080: first defined here
C:\Users\ASUS-PC\AppData\Local\Temp\build5893365334162908593.tmp\SoftwareSerial\SoftwareSerial.cpp.o: In function `SoftwareSerial::read()':
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial/SoftwareSerial.cpp:392: multiple definition of `__vector_4'
C:\Users\ASUS-PC\AppData\Local\Temp\build5893365334162908593.tmp\sketch_aug13a.cpp.o:C:\Program Files (x86)\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:473: first defined here
C:\Users\ASUS-PC\AppData\Local\Temp\build5893365334162908593.tmp\SoftwareSerial\SoftwareSerial.cpp.o: In function `SoftwareSerial::read()':
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial/SoftwareSerial.cpp:392: multiple definition of `__vector_5'
C:\Users\ASUS-PC\AppData\Local\Temp\build5893365334162908593.tmp\sketch_aug13a.cpp.o:C:\Program Files (x86)\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:473: first defined here
collect2.exe: error: ld returned 1 exit status
Error compiling.

so I think I would have to give up whole port where I am going to use SoftwareSerial.. such that Enable interrupt will not be usable to any pins on that port??? am I correct?

GreyGnome, 2015-08-17:
The short answer: Yes. Both libraries try to define the same interrupts, which causes a conflict. There is another issue here that covers it more indepth. I will look for it.x

GreyGnome, 2015-08-17:
See https://github.com/GreyGnome/EnableInterrupt/issues/7

GreyGnome, 2015-10-02:
@shirish47
I will close this issue. I am writing an FAQ page which will describe the problem and what to do.

closed 2015-10-02 by GreyGnome