GreyGnome/EnableInterrupt #18: Hi-Speed examples not compiling

engwan, 2015-08-18:
Hi, I'm not sure where to ask this but here's my problem:

I'm unable to compile any sketches when hi speed mode is enabled, including the examples.

Here's the compiler error:

```
In file included from /home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h:829:0,
                 from HiSpeedTest.ino:52:
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h: In function ‘void enableInterrupt(uint8_t, interruptFunctionType, uint8_t)’:
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:4:3: error: ‘risingPinsPORTB’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:9:3: error: ‘risingPinsPORTC’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:14:3: error: ‘risingPinsPORTD’ was not declared in this scope
In file included from /home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h:843:0,
                 from HiSpeedTest.ino:52:
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:22:3: error: ‘fallingPinsPORTB’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:27:3: error: ‘fallingPinsPORTC’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:32:3: error: ‘fallingPinsPORTD’ was not declared in this scope
In file included from /home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h:866:0,
                 from HiSpeedTest.ino:52:
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:44:3: error: ‘portSnapshotB’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:55:3: error: ‘portSnapshotC’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:66:3: error: ‘portSnapshotD’ was not declared in this scope
In file included from /home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h:957:0,
                 from HiSpeedTest.ino:52:
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h: In function ‘void disableInterrupt(uint8_t)’:
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:78:3: error: ‘risingPinsPORTB’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:79:3: error: ‘fallingPinsPORTB’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:86:3: error: ‘risingPinsPORTC’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:87:3: error: ‘fallingPinsPORTC’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:94:3: error: ‘risingPinsPORTD’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/utility/ei_PinChange328.h:95:3: error: ‘fallingPinsPORTD’ was not declared in this scope
In file included from HiSpeedTest.ino:52:0:
/home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h: In function ‘void __vector_3()’:
/home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h:1269:19: error: ‘portSnapshotB’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h:1270:19: error: ‘risingPinsPORTB’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h:1271:19: error: ‘fallingPinsPORTB’ was not declared in this scope
In file included from HiSpeedTest.ino:52:0:
/home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h: In function ‘void __vector_4()’:
/home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h:1328:19: error: ‘portSnapshotC’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h:1329:19: error: ‘risingPinsPORTC’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h:1330:19: error: ‘fallingPinsPORTC’ was not declared in this scope
In file included from HiSpeedTest.ino:52:0:
/home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h: In function ‘void __vector_5()’:
/home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h:1380:19: error: ‘portSnapshotD’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h:1381:19: error: ‘risingPinsPORTD’ was not declared in this scope
/home/engwan/sketchbook/libraries/EnableInterrupt/EnableInterrupt.h:1382:19: error: ‘fallingPinsPORTD’ was not declared in this scope
```

GreyGnome, 2015-08-20:
This is the right spot. I'll take a look- sorry about the issues!

GreyGnome, 2015-08-21:
I committed fixes to 0.8.1. It is on this github site only, and has not been tested. So you can grab the EnableInterrupt.h file immediately and test. I will test this weekend.

MakkyB, 2015-08-22:
I have the same problem. New file is not working for me, compiling is just hanging for about 15minutes now, atleast no errors :)

GreyGnome, 2015-08-24:
Can you grab the latest EnableInterrupt.h file from Github and replace your
current file, then give it a try?

Thanks.

GreyGnome, 2015-08-25:
Please send me an error report on this so I can mark it closed. Thanks.

juanricos, 2015-08-26:
Hello. I am having this same issue, even with latest EnableInterrupt.h. Here is my error output.

This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
Arduino: 1.0.6 (Windows 7), Board: "Arduino Mega 2560 or Mega ADK"
In file included from F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h:1011,
                 from HiSpeed.ino:38:
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h: In function 'void enableInterrupt(uint8_t, void (*)(), uint8_t)':
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:4: error: 'risingPinsPORTB' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:9: error: 'risingPinsPORTJ' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:14: error: 'risingPinsPORTK' was not declared in this scope
In file included from F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h:1029,
                 from HiSpeed.ino:38:
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:22: error: 'fallingPinsPORTB' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:27: error: 'fallingPinsPORTJ' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:32: error: 'fallingPinsPORTK' was not declared in this scope
In file included from F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h:1056,
                 from HiSpeed.ino:38:
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:44: error: 'portSnapshotB' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:55: error: 'portSnapshotJ' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:69: error: 'portSnapshotK' was not declared in this scope
In file included from F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h:1153,
                 from HiSpeed.ino:38:
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h: In function 'void disableInterrupt(uint8_t)':
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:81: error: 'risingPinsPORTB' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:82: error: 'fallingPinsPORTB' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:90: error: 'risingPinsPORTJ' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:91: error: 'fallingPinsPORTJ' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:98: error: 'risingPinsPORTK' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/utility/ei_PinChange2560.h:99: error: 'fallingPinsPORTK' was not declared in this scope
In file included from HiSpeed.ino:38:
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h: In function 'void __vector_9()':
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h:1463: error: 'portSnapshotB' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h:1464: error: 'risingPinsPORTB' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h:1465: error: 'fallingPinsPORTB' was not declared in this scope
In file included from HiSpeed.ino:38:
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h: In function 'void __vector_10()':
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h:1624: error: 'portSnapshotJ' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h:1625: error: 'risingPinsPORTJ' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h:1626: error: 'fallingPinsPORTJ' was not declared in this scope
In file included from HiSpeed.ino:38:
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h: In function 'void __vector_11()':
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h:1670: error: 'portSnapshotK' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h:1671: error: 'risingPinsPORTK' was not declared in this scope
F:\Arduino Libs and Sketches\Sketches\1.0.6\libraries\EnableInterrupt/EnableInterrupt.h:1672: error: 'fallingPinsPORTK' was not declared in this scope

GreyGnome, 2015-08-26:
Please download the .h file from the Github page and try it. Let me know
how it works. Thanks.

juanricos, 2015-08-27:
Yes that was with latest EnableInterrupt.h from August 20, 2015. I downloaded latest Arduino 1.6.5r2 to see if  that would make a difference but still cannot compile. 1.6.5 gives a little more info in error output.

```
Arduino: 1.6.5 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:1011:0,
                 from HiSpeed.ino:38:
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h: In function 'void enableInterrupt(uint8_t, interruptFunctionType, uint8_t)':
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:4:3: error: 'risingPinsPORTB' was not declared in this scope
   risingPinsPORTB |= portMask;
   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:9:3: error: 'risingPinsPORTJ' was not declared in this scope
   risingPinsPORTJ |= portMask;
   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:14:3: error: 'risingPinsPORTK' was not declared in this scope
   risingPinsPORTK |= portMask;
   ^
In file included from C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:1029:0,
                 from HiSpeed.ino:38:
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:22:3: error: 'fallingPinsPORTB' was not declared in this scope
   fallingPinsPORTB |= portMask;
   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:27:3: error: 'fallingPinsPORTJ' was not declared in this scope
   fallingPinsPORTJ |= portMask;
   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:32:3: error: 'fallingPinsPORTK' was not declared in this scope
   fallingPinsPORTK |= portMask;
   ^
In file included from C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:1056:0,
                 from HiSpeed.ino:38:
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:44:3: error: 'portSnapshotB' was not declared in this scope
   portSnapshotB=*portInputRegister(portNumber);
   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:55:3: error: 'portSnapshotJ' was not declared in this scope
   portSnapshotJ=*portInputRegister(portNumber);
   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:69:3: error: 'portSnapshotK' was not declared in this scope
   portSnapshotK=*portInputRegister(portNumber);
   ^
In file included from C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:1153:0,
                 from HiSpeed.ino:38:
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h: In function 'void disableInterrupt(uint8_t)':
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:81:3: error: 'risingPinsPORTB' was not declared in this scope
   risingPinsPORTB &= ~portMask;
   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:82:3: error: 'fallingPinsPORTB' was not declared in this scope
   fallingPinsPORTB &= ~portMask;
   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:90:3: error: 'risingPinsPORTJ' was not declared in this scope
   risingPinsPORTJ &= ~portMask;
   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:91:3: error: 'fallingPinsPORTJ' was not declared in this scope
   fallingPinsPORTJ &= ~portMask;
   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:98:3: error: 'risingPinsPORTK' was not declared in this scope
   risingPinsPORTK &= ~portMask;
   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/utility/ei_PinChange2560.h:99:3: error: 'fallingPinsPORTK' was not declared in this scope
   fallingPinsPORTK &= ~portMask;
   ^
In file included from HiSpeed.ino:38:0:
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/EnableInterrupt.h: In function 'void __vector_9()':
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:1463:19: error: 'portSnapshotB' was not declared in this scope
   changedPins   = portSnapshotB ^ current;
                   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:1464:19: error: 'risingPinsPORTB' was not declared in this scope
   tmp           = risingPinsPORTB & current;
                   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:1465:19: error: 'fallingPinsPORTB' was not declared in this scope
   interruptMask = fallingPinsPORTB & ~current;
                   ^
In file included from HiSpeed.ino:38:0:
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/EnableInterrupt.h: In function 'void __vector_10()':
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:1624:19: error: 'portSnapshotJ' was not declared in this scope
   changedPins   = portSnapshotJ ^ current;
                   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:1625:19: error: 'risingPinsPORTJ' was not declared in this scope
   tmp           = risingPinsPORTJ & current;
                   ^
[... cut ...]
   changedPins   = portSnapshotK ^ current;
                   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:1671:19: error: 'risingPinsPORTK' was not declared in this scope
   tmp           = risingPinsPORTK & current;
                   ^
C:\Users\Jason\Documents\Arduino\libraries\EnableInterrupt/EnableInterrupt.h:1672:19: error: 'fallingPinsPORTK' was not declared in this scope
   interruptMask = fallingPinsPORTK & ~current;
                   ^
Error compiling.
```

GreyGnome, 2015-08-28:
Ach, sorry- try it now.

juanricos, 2015-08-28:
What to try? What file should I update? I still only see EnableInterrupt.h from August 20, 2015.

juanricos, 2015-08-28:
Sorry. I did not see 0.8.2. It is working now. Thank you much.

GreyGnome, 2015-09-30:
Ok great. I was having Git problems so I did not put up the proper file. But anyway the 0.8.2 library has it fixed.

closed 2015-09-30 by GreyGnome