GreyGnome/EnableInterrupt #20: This library cannot be used in bigger projects.

Oligodendrogliom, 2015-08-30:
If this library is used in projects spread over more than a single INO-file, i.e. if separate headers and code files are used, it breaks the linking process. 

Because all code is simply put into a header file, the linker will produce a load of errors concerning multiple definitions. This cannot be catched by your include guards. For working in more than simple projects, this libray must be split into a separate code file (.cpp) and a header file containing solely declarations. I had the same problems already with the predecessor PinChangeInt.

GreyGnome, 2015-08-31:
Why was this closed? Can you send me some code examples? I am surprised to see you say that it cannot be caught by the include guards. I would like to see how it can be split into a separate code file and a header file, too.

Thanks.

DanielOgorchock, 2015-08-31:
I don't have any solutions, but I have experienced the same issue with this library. If I try to use EnableInterrupt in my own library, the build fails.

stickbreaker, 2015-09-04:
I use this library to implement CTS/RTS handshaking.  I have modified HardwareSerial, a core library of the Arduino system.  I also use this library for interrupt services for Keyboard, Timeclock, and Limit Switch interrupts.  Since HardwareSerial is included 99% of the time, I just use the following definition for other Libraries or usage from the 'main' application.  
~~~c++
extern void enableInterrupt(uint8_t interruptDesignator, void (*userFunction)(void), uint8_t mode);
extern void disableInterrupt(uint8_t interruptDesignator);
extern uint8_t interruptedArduinoPin;
~~~
chuck.

GreyGnome, 2015-09-15:
@DanielOgorchock Please send me a code sample so I can try to compile. Thanks.

stickbreaker, 2015-09-15:
@GreyGnome , 
   To simulate the problem he is having, write a library that uses your *EnableInterrupt* library, include and use *EnableInterrupt* in another library, include and use *EnableInterrupt* in your main application sketch.  You will see the 'redefined Vector' problem.  I solved it by not including *EnableInterrupt* in any of my libraries or Applications, except HardwareSerial.cpp.
Chuck

GreyGnome, 2015-10-02:
@DanielOgorchock  @stickbreaker 
Yes, the EnableInterrupt library code can only be compiled once per project. To prevent this from happening, all files that include the EnableInterrupt.h need to have the following prior to its include:
```
#define LIBCALL_ENABLEINTERRUPT
```
There should be one file and only one file per all the files compiled for a sketch that includes EnableInterrupt.h but does not include that #define. Then the code will get compiled, and the linker will find only the single definition, and it will link all the object files' references to it.

If a file doesn't reference any EnableInterrupt component then of course there is no issue; just don't `#include EnableInterrupt.h`

I will close this issue. It may get messy but ultimately the LIBCALL_ENABLEINTERRUPT should be able to reliably prevent this issue.

closed 2015-10-02 by GreyGnome