GreyGnome/EnableInterrupt #21: Multiple pins detected as 1 PCInt

lemontini, 2015-09-07:
I am using DS3231 RTC with my Arduino Yun, and connect DS3231's SQW pin (generates interrupt at preset alarm time) to arduino pin 10. EnableInterrupt detects the interrupt, but it also detects it when I connect the wire to pin 9. It also detects it occasionally when I connect it to pin 11 - though it seems to be more like random. I am concerned about this because pins 9, 11 are supposed to control different operations in the future of my project.
What could be done? Should I better use A0-A7 pins instead?

GreyGnome, 2015-09-07:
Could you make sure the pins are connected and then try again? That is,
make sure that every pin in question has a connection to either +V or
Ground. Then apply to SQW signal to the other pins, and see what happens.

I test every pin on my 328 and 2560 but I don't have a Yun, so let's hope I
didn't screw that one up... :-\

lemontini, 2015-09-08:
I was testing it with all other pins but pin 10 disconnected (of course I used 5V and GND pins and then SCA and SCL for communication, which are the ones on the top to the left of the AREF pin, see in this picture: https://yaler.net/fritzing-arduino-yun.png). I'd noticed that I get an alarm on pin 10, I was happy, but then I'd decided to probe other ones, which gave me a surprise.
Yesterday I haven't got a chance to test if it interferes with my other setup connected, but I will try that this evening and I'll try to update You.
As of A0-A5 - I've noticed they aren't even supported as Your manual says (Leonardo, cause it should be compatible with Yun), so they were not giving any response, which is understandable.

GreyGnome, 2015-09-08:
Thanks let me know. I'm concerned that a floating pin may be interfering
with things somehow.

Also, could you send me your sketch? I'd like to look at it and make sure
all interrupt code is in order.

Thanks.

stickbreaker, 2015-09-09:
Did @lemontini say the other pins were floating? Unless they are pulled up or pulled down, It would not surprise me that he is seeing false interrupts.

GreyGnome, 2015-09-10:
I'm not sure. But I want to make sure they're not, that's for sure! :-)

GreyGnome, 2015-09-30:
@lemontini Do you have any updates on this issue? If it's still broken, please send me your sketch so I can check it out.

lemontini, 2015-09-30:
Sorry to not update on the issue for so long. The problem is I disconnected the DS3231 from my setup in favor of using native Yun's capability of time syncing from the linux side via NTP - it was because it somehow interfered with my other setup - with PWM control of the 12V 4-wire PC Fan (which was controlled with the help of interrupt timers, which had to be upscaled to 25 kHz).
The other problem I faced when I excluded the DS3231 out of the equation - my temperature and humidity sensor DHT22 tended to become hung after some random periods of readings (it could be reset by cutting and restoring power to its Vcc pin) which I found to be not my only personal problem - some people on the net seem to be experiencing it also.

About getting back to the point - when I still had DS3231 connected and was testing the setup, I WAS exercising some strange behavior of pin floating - it actually made some erroneous interrupt interceptions. But judging from all stated above, it seems that all problems tend to arise from PWM control of the fan - i.e. with tinkering with hardware timers, which could mess the timing functions of Yun, which directs me to the guessing that it could also mess the functionality of EnableInterrupt library functions. Because - let's say I had stripped the setup down to only two versions:
(1) the DHT22 sensor and
(2) DHT22 + PWM fan control
and in the (2) setup the DHT22 hanging was also occurring, while in the (1) it seemed to read the values without random hanging.

So, the bottom line is: can someone help to solve the problem? I need to have the two elements of the (2) setup in my sketch - that is the main axis of the project. Which is completely off the scope of the EnableInterrupt library, though I tend to think that if the said problem would be solved, there might not be any problems with the pin floating, so... :)

GreyGnome, 2015-10-02:
EnableInterrupt uses completely different interrupt vectors than the ATmega timers, so it should not be affected. As a matter of fact, the pin interrupts take priority over the timer interrupts so unless you spend a lot of time in one of the interrupts (which is entirely possible), the pin interrupts will take priority.

I understand you want to have 2 DHT22's and 2 PWM fan controllers on your Yun. First thing is to get one working 100% and then add the second.

Can you set it up that way and get some reproducible results? The broader the problem set the harder it is to debug. First thing is to simplify, then verify, then expand until you have a completely working system. Once you have a functioning problem (an oxymoron, I'll admit): the sketch + the components- then upload the sketch here, tell me what is connected to what, and we'll take a look.

GreyGnome, 2015-10-19:
Are you still having issues? If so I'd like to take a look. If not I'd like to close this issue.

lemontini, 2015-10-19:
maybe it would be better to close it, since currently I have no spare time
for testing, sorry ;)

GreyGnome, 2015-10-20:
Ok, if you bump into this in the future feel free to reopen. In the meantime I will probably get a 32u4 and test with it sometime.

closed 2015-10-20 by GreyGnome