GreyGnome/EnableInterrupt #22: Sleep Modes

shirish47, 2015-09-15:
How Do I use Sleep modes with this library?

I am thinking of using lowpower library

GreyGnome, 2015-09-15:
That's a little open-ended and broad, isn't it? Perhaps you could read this: http://stackoverflow.com/help/how-to-ask . I don't have any expertise with ATmega sleep modes and can't give you a primer in using sleep; there are many articles that will show you how to do it. Here is an intro from the Arduino playground: http://playground.arduino.cc/Learning/ArduinoSleepCode . Note that you can almost exactly replace attachInterrupt() detachInterrupt() with enableInterrupt() disableInterrupt() but use the actual Arduino pin instead of the "Interrupt number".

shirish47, 2015-09-15:
:) thanks ... ya I tried other library ..Lowpower library.. but as usual ..it conflicts with the vectors defined in this one. anyways I would go program for sleep with registers.. hope things to work.

GreyGnome, 2015-09-30:
I will close this. But check the Wiki page here... you can comment out vectors in the EnableInterrupt library. You may be able to get it to work.

closed 2015-09-30 by GreyGnome