GreyGnome/EnableInterrupt #23: License?

molecule, 2015-09-22:
I'd be very interested in using your library in a project. Would it be possible for you to add a license to this library? Also would you be able to do LGPL, or BSD, or something similar? My company may not allow me to use this library if it's Apache 2.0.

Thanks so much for your consideration.

GreyGnome, 2015-09-22:
The license is already Apache 2.0 as it says at the top of the
EnableInterrupt.h file.

What is the objection to Apache 2.0? I think it's quite liberal. That said,
I am happy to accommodate as I can't imagine a more liberal license (except
for the WTFPL) but I'd like to know what's up before I sign off.

I'm not sure what the LGPL is all about but I think it's even more
restrictive than Apache 2.0, in which case I don't think I'd mind but again
I'd like to know what's up. Thanks.

molecule, 2015-09-25:
Sorry for the delay. I've scheduled a meeting with the company's lawyers this afternoon to discuss the issues. I was also under the impression that the Apache 2.0 license was friendly, so I will let you know what I find out.

GreyGnome, 2015-09-30:
I'll close this issue. Send me a message if you need any more assistance.

closed 2015-09-30 by GreyGnome

molecule, 2015-09-30:
Hi Mike,

Sorry about the delay. I did speak with the lawyers. Apparently Apache 2.0
could possibly be interpreted as giving license to every patent in the
company, and that makes the lawyers nervous. They prefer Clear BSD or MIT
licenses. Personally, I understand the value of a license like Apache 2.0,
and I can't tell you what I would do. All I can say is: I can't use any
library that has Apache 2.0, but I CAN use Clear BSD or MIT licenses. So,
feel free to take that into account as you consider changing licenses.

Thanks so much for your time,
Molly

GreyGnome, 2015-10-07:
Oy vay... Do your lawyers mean to tell me, that after all the years of the
Apache license, they think that it can possibly be interpreted in such a
way, and that no one else noticed or has expressed any such concern? Do
they think that that would be even acceptable to the Open Source community?
That's a horrible time bomb. Geez Loueez, ridiculous. If that was possible
I would bail from the Apache license straightaway. I actually chose it
*because* it's permissive, yet I hate most software patents and so I like
its clause about patents. But the license only and specifically applies to
the "Work", where (and I quote) ' "Work" shall mean the work of authorship,
whether in Source or Object form, made available under the License, as
indicated by a copyright notice that is included in or attached to the
work...' Since your company's software is not made available under the
License, the Grant of Patent License clause does not apply to it.

So I'm not willing to change the license on the library for general
distribution.

Now that I've got that off my chest :-), as the sole copyright owner I am
willing to license it to your company in terms that they feel more
comfortable with, provided that they include a Grant of Patent License
substantially similar to the clause contained in the Apache 2.0 license
(which would of course be as explicit as necessary to make clear that it
applies only to the EnableInterrupt library, and not any of their work that
utilizes the library). Your company's rights would continue through future
versions of the EnableInterrupt library, so you could use it whatever
future Arduino chips come along. Furthermore, you would not need to
distribute a copy of our agreement with your product; may main concern is
attribution. So I'm thinking the Apache 2.0 license can be used as
boilerplate for a modified agreement.

If you wish, why don't we take this conversation offline? Contact me at
mschwage @ gmail . com .

molecule, 2015-10-07:
Haha....I don't know what to say. I'm not in a position to argue for or against their interpretation. I'm also not interested in defending their stance, or negotiating a special license. Thanks for the offer. 

I found another solution that does not involve your library. 

Best of luck,
Molly