GreyGnome/EnableInterrupt #24: Support for Arduino Zero

q2dg, 2015-10-01:
I'm haven't this board so I can't see if this fantastic library is already compatible or not, but as anywhere is said it is, I'd like to know if this compatibility exists, and if it doesn't, ask for it.
Thanks!

GreyGnome, 2015-10-02:
The Arduino Zero uses the same model as the Due: Every pin can service an interrupt, and the interrupt model is much different from the ATmega series: Rather than requiring an "interrupt number" as an argument to attachInterrupt(), you simply give it a pin number. As a matter of fact, the library's support of the Due is embarrassingly simple:
```
#if defined __SAM3U4E__ || defined __SAM3X8E__ || defined __SAM3X8H__
#define enableInterrupt(pin,userFunc,mode) attachInterrupt(pin, userFunc,mode)
#define disableInterrupt(pin) detachInterrupt(pin)
```
I simply need to find what CPU the Zero uses... how it's defined by the compiler, that is- and, voila'! It's supported.

If you want to support interrupts on the Zero, then, you could just use attachInterrupt()/detachInterrupt(). If you want your code to be compatible with the ATmega series then the EnableInterrupt library is a value-add. Otherwise you could ignore it.

But if you *like* it- just because- then yes I will add Zero 'support' to the library. :-)

q2dg, 2015-10-03:
Ooh, ok, I understand now.
Thanks a lot for the clarification.
I'll close the issue, then, if you don't mind.
Thanks!

closed 2015-10-03 by q2dg

GreyGnome, 2015-10-07:
In the ensuing weeks I will make sure the library compiles on the Zero.