GreyGnome/EnableInterrupt #25: Examples won't compile

RobertClemenzi, 2015-10-13:
I have tried 3 examples, and none of them compiles. I am using Arduino 1.0.5 IDE. The following are just examples, there are many more errors.

EnableInterrupt OOSimple
OOSimple.ino: In function 'void* createSimple()':
OOSimple:10: error: previous declaration of 'void* createSimple()' with 'C++' linkage

EnableInterruptTest:183: error: 'ByteBuffer' does not name a type
EnableInterruptTest.ino: In function 'void quicfunc0()':
EnableInterruptTest:228: error: 'PCintPort' has not been declared

SimpleWithLibrary.cpp.o: In function `UselessClass':
/Useless.h:19: undefined reference to `UselessClass::init(unsigned char, unsigned char)'

In my own code, when I include the header file in just my library cpp file, I get
C:\...\mcHC_SR04.cpp:81: error: 'disableInterrupt' was not declared in this scope
To make the code work, I had to include the header in both the main sketch and the library file. The #define LIBCALL_ENABLEINTERRUPT worked in either file.

GreyGnome, 2015-10-15:
Thanks, I'm working on it. I have the OOSimple working now.

GreyGnome, 2015-10-15:
EnableInterruptTest will not work. That's for internal use only. SimpleWithLibrary works for me; can you make sure that there are Useless files in the directory (folder)?

Your last paragraph is correct; that's how it works. Let me know if you don't understand something about it. If a file references something from the EnableInterrupt library, you will need to include it.

GreyGnome, 2015-10-15:
I will exclude examples that should not be included in the distribution, from the distribution's zip file. Even better, I will create a "Tests" directory.

RobertClemenzi, 2015-10-15:
They are both in the same directory.

C:\Users\[xxx]\Documents\Arduino\libraries\EnableInterrupt\examples\SimpleWithLibrary\Useless.h
C:\Users\[xxx]\Documents\Arduino\libraries\EnableInterrupt\examples\SimpleWithLibrary\SimpleWithLibrary.ino

However, the private *init* method is defined after it is used.

RobertClemenzi, 2015-10-15:
>"If a file references something from the EnableInterrupt library, you will need to include it."

So, that is why I included it in my cpp library file, but why is it also required in the ino (sketch) file? I never reference your library from there. The ino file references my library file and only that file references your library. Yet, both files must include a reference to your library to compile.

GreyGnome, 2015-10-15:
It appears as if the Arduino system does not know to include the library's path unless you include it in the main sketch. I will add this to the FAQ.

GreyGnome, 2015-10-16:
Let me know if I have missed something. 0.9.1 is out now, which fixes the OOsimple issue. If we're all good I would like to close this issue.

RobertClemenzi, 2015-10-17:
I have verified that OOsimple compiles and that EnableInterruptTest is no longer available.

However, SimpleWithLibrary.ino still fails to compile. The problem is that most of the methods do not have implementations. (I checked this by adding null implementations to them - empty curly braces - and the sketch compiled.) Perhaps the cpp file is missing.

RobertClemenzi, 2015-10-18:
I found the problem - Useless.cpp is where it should be on github but is missing from the 0.9.0 and 0.9.1 zip files.

GreyGnome, 2015-10-19:
Fixed.  Thanks!

closed 2015-10-19 by GreyGnome