GreyGnome/EnableInterrupt #26: arduinoInterruptedPin variable not available in my library files

RobertClemenzi, 2015-11-02:
When debugging code I placed in a library, the arduinoInterruptedPin variable would have been a big help, but was not available. The reason is that variables can not be defined in header files that are included in multiple source files. When they are, a **multiple definition** error is generated. The <b>EnableInterrupt</b> library is implemented as a header file with 2 sections, 
 * One that is included in all the files that use the library - the header section, and 
 * One that must be included in only the top level sketch (source file) - the implementation section

My suggested solution is to 
 * Move the variable definition to the implementation section
 * Add a function prototype to the header section
 * Implement the function in the implementation section

In the Arduino C implementation, function prototypes can be included via a header file in multiple source files without any problem. The function should simply return the value of the variable. This modification should allow all existing code that uses the variable to continue operating without any changes, and will allow libraries to also access the interrupting pin numbers.

The following is code that I have tested. You may use it if you like, or change what ever makes sense. I am not trying to tell you what to do, just making a suggestion.

```
#ifdef EI_ARDUINO_INTERRUPTED_PIN
uint8_t arduinoInterruptedLastPin();      // function prototype
#endif

#ifndef LIBCALL_ENABLEINTERRUPT           // This is the magic dividing line
                                          // between prototypes and implementation
#ifdef EI_ARDUINO_INTERRUPTED_PIN
volatile uint8_t arduinoInterruptedPin=0; // This was moved from above the line

uint8_t arduinoInterruptedLastPin(){      // function implementation
  return arduinoInterruptedPin;
}
#endif
```
I also defined the variable as **volatile** because it is set via an interrupt routine.

GreyGnome, 2015-11-03:
Thanks, Robert. I will fix this.

GreyGnome, 2015-11-03:
By making arduinoInterruptedPin volatile, any part of the sketch can simply reference it. The function is unnecessary in my view. Let me know if I don't understand something about that.

Regarding the variable, in the latest code I have fixed it by defining the variable in the #ifndef LIBCALL_ENABLEINTERRUPT. Then I added a #else clause and declared the variable there. Like this:
```
#ifndef LIBCALL_ENABLEINTERRUPT
#ifdef EI_ARDUINO_INTERRUPTED_PIN
volatile uint8_t arduinoInterruptedPin=0;
#endif
...(a lot of library code here)...
#else
#ifdef EI_ARDUINO_INTERRUPTED_PIN
extern volatile uint8_t arduinoInterruptedPin;
#endif
#endif
```

RobertClemenzi, 2015-11-06:
I tested your solution and it works.

I think it would be easier for others to follow if you keep the extern declaration in the header section instead of placing it at the end, but that is just one opinion.

GreyGnome, 2015-11-07:
0.9.4 is being uploaded as I speak.

GreyGnome, 2015-11-09:
I am going to close this. The change you suggested is in the source file and will be included in the next release, so I don't think we need to leave this issue open. Thanks for the bug report!

closed 2015-12-05 by GreyGnome