GreyGnome/EnableInterrupt #27: Extra comma in const array declaration

RobertClemenzi, 2015-11-03:
\EnableInterrupt/EnableInterrupt.h:193: warning: only initialized variables can be placed into program memory area
```
const uint8_t PROGMEM digital_pin_to_port_bit_number_PGM[] = {
  0, // 0 == port D, 0
  1,
  2,
.......
  4,
  5,
};
```
The comma after the last 5 should not be there.

GreyGnome, 2015-11-03:
Robert you are a one-man debugging machine! Thanks, I will fix this.

RobertClemenzi, 2015-11-06:
I just tested the obvious fix - I removed the comma.
However, that did not make the warning go away. Apparently, this is a known problem with the "PROGMEM" macro.

BTW, the extra commas are in every section where that variable is defined.

GreyGnome, 2015-11-07:
Thanks. Fixed in 0.9.4.

GreyGnome, 2015-11-07:
Strangely, I don't see the warning at all... I am using avr-g++ (Fedora 4.9.2-1.fc22) 4.9.2 . What version are you using?

RobertClemenzi, 2015-11-12:
I am using the Arduino 1.0.5 IDE which is associated with many versions of gcc. It appears that the current version is avr-gcc-4.3.2.exe. However, I don't think the gcc version matters. Instead, the important files are 

  arduino-1.0.5-windows\arduino-1.0.5\hardware\tools\avr\avr\include\avr\pgmspace.h

which defines the PROGMEM macro and the 3 files similar to

  arduino-1.0.5-windows\arduino-1.0.5\hardware\tools\avr\libexec\gcc\avr\4.3.2\cc1.exe

which contain the text of the warning. In order to see the warning in the IDE I am using, from the menu, select

  File / Preferences / Show verbose output during compilation

Additional info is at

   http://forum.arduino.cc/index.php/topic,85840.0.html

GreyGnome, 2015-12-05:
As this is a known problem with PROGMEM and (regardless) I removed the spurious comma, I am closing this issue.

closed 2015-12-05 by GreyGnome