GreyGnome/EnableInterrupt #28: Testing EnableInterrupt on ATmega1284p

Cat30Mk3, 2015-11-18:
Hi Mike,
I've been testing the Simple example on a LowPower labs Moteino Mega (ATMEGA1284p) .
I believe I may have encountered a defect in the EnableInterrupt  library I downloaded on Nov 16 2015.

My tests using a single pushbutton were only partially successful. I was able to run the Simple example successfully on pins 0 to 7, A0 to A7, and 10 and 11 ... but was unable to activate pin change interrupts on pins 12 through 23. I have confirmed that the hardware and wiring is functioning correctly.

I'd be happy to work with you to confirm and resolve. 

 I've run your Mighty1284p example, but I'm afraid I don't understand the expected outcome.
I am happy to help if you give me a little guidance.

John 
john.witherspoon@sympatico.ca

GreyGnome, 2015-11-19:
Hi,
Could you tell me how you're compiling this? It sounds like maybe the proper configuration is not being loaded... where did you get your boards.txt file from, etc.?

Cat30Mk3, 2015-11-19:
Hi Mike, you've posed an interesting question...

I'm running Arduiino IDE 1.0.6 because Felix at LowPower Labs mentioned it is his preference over  1.6.4  
And I'm using his Moteino Core files from his Github repository  that include the board.txt and pins_arduino.h. They are located under the hardware folder of my main Arduino1.0.6 folder. 
Before we go any further, I think I should check with Felix to make sure I have installed the right version of the core files  and confirm which IDE version he recommends.
Is there any compatibility issue with your library and 1.0.6 Arduino IDE version? What version of Arduino IDE  do you recommend?

John

Cat30Mk3, 2015-11-20:
Mike, just to be certain I am testing on a clean platform, tonight I downloaded and installed the Arduino IDE latest version (1.6.6) on a laptop running Windows 7.

 I then followed the LowPowerLabs instructions to download and install their latest MoteinoMega core files from the LowPowerLabs github site into the hardware directories:
C:\Program Files (x86)\Arduino\hardware\Moteino\avr ...
 and 
C:\Users\Linda\Documents\Arduino\hardware\Moteino\avr..
These folders contain the boards.txt and pins_arduino.h files (I have attached them below)

[pins_arduino.h.txt](https://github.com/GreyGnome/EnableInterrupt/files/39682/pins_arduino.h.txt)
[boards.txt](https://github.com/GreyGnome/EnableInterrupt/files/39683/boards.txt)

Then I downloaded your latest EnableInterrupt library from your github site and copied it to the libraries directory and renaming to drop the '-Master' suffix.
Running the Arduino IDE I selected the MoteinoMega Board under the tools menu, and the Simple example from the EnableInterrupt examples under the File>Examples menu. The program compiled and downloaded without errors or warnings. 

The physical setup is a new MoteinoMega board connected via an Adafruit FTDI Friend board to the laptop USB (Com9) with a single tactile pushbutton connecting from the MoteinoMega GND pin to each data pin - one at a time as I edit the #define ARDUINOPIN preprocessor definition in the Simple.ino example program  and recompile. 

** I found the tests resulted in the same failures.**

Here are the test results of this setup:

/*test results 19/11/15 20:30
* Simple.ino MoteinoMega (No transciever) pushbutton 'pin to gnd' 
Pass Pins: 7,6,5,4,3,2,1,0,A0,A1,A2,A3,A4,A5,A6,10,11
Fail Pins:12,13,14,15,16,17,18,19,20,21,22,23
Not Tested Pins: 8,9 (in use for Serial Monitor communications)
*/

John

GreyGnome, 2015-11-20:
Ok, awesome testing. It seems like a bug in the software, for sure. Let me run through it and figure out a couple of debug steps we can perform. ...Or maybe I'll find the problem :-)

GreyGnome, 2015-11-20:
Ok, here's the difference between digital pins 7 and 12, for example: I compiled the program using the Mighty-1284p files. After each compile, I dumped the output to assembler. Here's an example for pin 12:
```
avr-objdump -S build-mighty_opt/Simple.elf > pin12.asm
```
Can you find the elf file and dump it, as I've shown, for both a working and non-working pin number? Now I rather suspect some other incompatibility, as the code looks decent to me and the only difference in the generated assembler is the pin number, as you can see here.
```
1446c1446
<      860:     87 e0           ldi     r24, 0x07       ; 7
---
>      860:     8c e0           ldi     r24, 0x0C       ; 12
1452c1452
<      86c:     87 e0           ldi     r24, 0x07       ; 7
---
>      86c:     8c e0           ldi     r24, 0x0C       ; 12
```
The pin sets (working and not) are interesting, though, because they are basically Pin Change Interrupt ports C and D. Pins 8 and 9 (in port D) you're not testing. Pins 10 and 11 (port D again) are selected to be External interrupts. The rest of the pins are exactly the body of pins covered by ports D and C for Pin Change Interrupts.

I wonder if you could do some debug printf's? In the EnableInterrupt.h file there is a line:
```
portBitNumber=pgm_read_byte(&digital_pin_to_port_bit_number_PGM[arduinoPin]);
```
Immediately following that line, could you print the port bit numbers? Let's stick to pins 7 and 12 for now, as I have done.

Next, could you put
```
extern volatile uint16_t interruptCount;
```
Right at the beginning of the EnableInterrupt.h file, and then right below this line:
```
  portSnapshotD = current;
```
put:
```
interruptCount++;
```
...We'll see if the interrupt count is actually being incremented in the ISR. This will tell us if we have an issue calling out to the user's subroutine.

Cat30Mk3, 2015-11-20:
Hi Mike,  I have been able to add the print and interruptCount variable as well as create the .asm files you requested. I retested all pins twice and tabulated the results since I found some different behaviour this morning. I cant explain it, but pins 12,13,14,15 are now generating interrupts - only pins 16 through 23 are failing. However only pins 10 through 15 are causing the new interruptCount variable to increase - for all other pins the new statement interruptCount++; is not executing. I ran one series of tests using "enableInterrupt(ARDUINOPIN | PINCHANGEINTERRUPT, interruptFunction, CHANGE);" and another set using " enableInterrupt(ARDUINOPIN, interruptFunction, CHANGE);"
In the latter case pins 2,10 and 11 demonstrated that the debug code you asked for was not executed -- I'm assuming this is the correct external interrupt default behaviour.

I'm also attaching pin7.asm, pin12.asm and pin20.asm (since pin12 now seems to be working but pin20 continues to fail).

/* debugging test results   20/11/15 09:10
* using enableInterrupt(ARDUINOPIN | PINCHANGEINTERRUPT, interruptFunction, CHANGE);
ARDUINOPIN 7,  arduinoPin=7,   portBitNumber=7, PASS, NO InterruptCount increments from .h
ARDUINOPIN 6,  arduinoPin=6,   portBitNumber=6, PASS, NO InterruptCount increments from .h
ARDUINOPIN 5,  arduinoPin=5,   portBitNumber=5, PASS, NO InterruptCount increments from .h
ARDUINOPIN 4,  arduinoPin=4,   portBitNumber=4, PASS, NO InterruptCount increments from .h
ARDUINOPIN 3,  arduinoPin=3,   portBitNumber=3, PASS, NO InterruptCount increments from .h
ARDUINOPIN 2,  arduinoPin=2,   portBitNumber=2, PASS, NO InterruptCount increments from .h
ARDUINOPIN 1,  arduinoPin=1,   portBitNumber=1, PASS, NO InterruptCount increments from .h
ARDUINOPIN 0,  arduinoPin=0,   portBitNumber=0, PASS, NO InterruptCount increments from .h
ARDUINOPIN A0, arduinoPin=24,  portBitNumber=0, PASS, NO InterruptCount increments from .h
ARDUINOPIN A1, arduinoPin=25,  portBitNumber=1, PASS, NO InterruptCount increments from .h
ARDUINOPIN A2, arduinoPin=26,  portBitNumber=2, PASS, NO InterruptCount increments from .h
ARDUINOPIN A3, arduinoPin=27,  portBitNumber=3, PASS, NO InterruptCount increments from .h
ARDUINOPIN A4, arduinoPin=28,  portBitNumber=4, PASS, NO InterruptCount increments from .h
ARDUINOPIN A5, arduinoPin=29,  portBitNumber=5, PASS, NO InterruptCount increments from .h
ARDUINOPIN A6, arduinoPin=30,  portBitNumber=6, PASS, NO InterruptCount increments from .h
ARDUINOPIN A7, arduinoPin=31,  portBitNumber=7, PASS, NO InterruptCount increments from .h
ARDUINOPIN 10, arduinoPin=10,  portBitNumber=2, PASS, InterruptCount increments from .h
ARDUINOPIN 11, arduinoPin=11,  portBitNumber=3, PASS, InterruptCount increments from .h
ARDUINOPIN 12, arduinoPin=12,  portBitNumber=4, PASS, InterruptCount increments from .h
ARDUINOPIN 13, arduinoPin=13,  portBitNumber=5, PASS, InterruptCount increments from .h
ARDUINOPIN 14, arduinoPin=14,  portBitNumber=6, PASS, InterruptCount increments from .h
ARDUINOPIN 15, arduinoPin=15,  portBitNumber=7, PASS, InterruptCount increments from .h
ARDUINOPIN 16, arduinoPin=16,  portBitNumber=0, FAIL, NO InterruptCount increment from .h
ARDUINOPIN 17, arduinoPin=17,  portBitNumber=1, FAIL, NO InterruptCount increment from .h
ARDUINOPIN 18, arduinoPin=18,  portBitNumber=2, FAIL, NO InterruptCount increment from .h
ARDUINOPIN 19, arduinoPin=19,  portBitNumber=3, FAIL, NO InterruptCount increment from .h
ARDUINOPIN 20, arduinoPin=20,  portBitNumber=4, FAIL, NO InterruptCount increment from .h
ARDUINOPIN 21, arduinoPin=21,  portBitNumber=5, FAIL, NO InterruptCount increment from .h
ARDUINOPIN 22, arduinoPin=22,  portBitNumber=6, FAIL, NO InterruptCount increment from .h
ARDUINOPIN 23, arduinoPin=23,  portBitNumber=7, FAIL, NO InterruptCount increment from .h
*/

/* debugging test results   20/11/15 09:10
* using enableInterrupt(ARDUINOPIN, interruptFunction, CHANGE);
ARDUINOPIN 7,  arduinoPin=7,   portBitNumber=7, PASS, NO InterruptCount increments from .h
ARDUINOPIN 6,  arduinoPin=6,   portBitNumber=6, PASS, NO InterruptCount increments from .h
ARDUINOPIN 5,  arduinoPin=5,   portBitNumber=5, PASS, NO InterruptCount increments from .h
ARDUINOPIN 4,  arduinoPin=4,   portBitNumber=4, PASS, NO InterruptCount increments from .h
ARDUINOPIN 3,  arduinoPin=3,   portBitNumber=3, PASS, NO InterruptCount increments from .h
ARDUINOPIN 2,  arduinoPin=2,   portBitNumber=2, PASS, NO InterruptCount increments from .h, portBitNumber=pgm_read_byte(.. NOT EXECUTED
ARDUINOPIN 1,  arduinoPin=1,   portBitNumber=1, PASS, NO InterruptCount increments from .h
ARDUINOPIN 0,  arduinoPin=0,   portBitNumber=0, PASS, NO InterruptCount increments from .h
ARDUINOPIN A0, arduinoPin=24,  portBitNumber=0, PASS, NO InterruptCount increments from .h
ARDUINOPIN A1, arduinoPin=25,  portBitNumber=1, PASS, NO InterruptCount increments from .h
ARDUINOPIN A2, arduinoPin=26,  portBitNumber=2, PASS, NO InterruptCount increments from .h
ARDUINOPIN A3, arduinoPin=27,  portBitNumber=3, PASS, NO InterruptCount increments from .h
ARDUINOPIN A4, arduinoPin=28,  portBitNumber=4, PASS, NO InterruptCount increments from .h
ARDUINOPIN A5, arduinoPin=29,  portBitNumber=5, PASS, NO InterruptCount increments from .h
ARDUINOPIN A6, arduinoPin=30,  portBitNumber=6, PASS, NO InterruptCount increments from .h
ARDUINOPIN A7, arduinoPin=31,  portBitNumber=7, PASS, NO InterruptCount increments from .h
[... cut ...]
ARDUINOPIN 18, arduinoPin=18,  portBitNumber=2, FAIL, NO InterruptCount increments from .h
ARDUINOPIN 19, arduinoPin=19,  portBitNumber=3, FAIL, NO InterruptCount increments from .h
ARDUINOPIN 20, arduinoPin=20,  portBitNumber=4, FAIL, NO InterruptCount increments from .h
ARDUINOPIN 21, arduinoPin=21,  portBitNumber=5, FAIL, NO InterruptCount increments from .h
ARDUINOPIN 22, arduinoPin=22,  portBitNumber=6, FAIL, NO InterruptCount increments from .h
ARDUINOPIN 23, arduinoPin=23,  portBitNumber=7, FAIL, NO InterruptCount increments from .h
*/

Cat30Mk3, 2015-11-20:
oops sent before I could clean up formatting and add .asms
The last set is same as first except for pins 2,10,11
Here are the .asms ...

[pin7.asm.txt](https://github.com/GreyGnome/EnableInterrupt/files/40417/pin7.asm.txt)
[pin11.asm.txt](https://github.com/GreyGnome/EnableInterrupt/files/40418/pin11.asm.txt)
[pin12.asm.txt](https://github.com/GreyGnome/EnableInterrupt/files/40419/pin12.asm.txt)
[pin20.asm.txt](https://github.com/GreyGnome/EnableInterrupt/files/40420/pin20.asm.txt)

GreyGnome, 2015-11-20:
With the exception of the pins that are failing to interrupt, all behavior is as expected. Only those interrupts on port D would increment that variable because of the ++ line that I asked you to include. 

Thanks for the check on the external interrupts. Yes, that is the expected behavior.

So now only pins on Port C appear to be failing. Could you take out the interruptCount++ statement that you put in the PORTD ISR, and add the statement after this statement:
```
  portSnapshotC = current;
```
...that is, include 
```
interruptCount++;
```
on the line immediately following it? Let's see what happens on the failing pins.

Thanks.

Weird that PortD now appears to be working... something strange seems to be going on... I'll review a couple of the ASM files.

Cat30Mk3, 2015-11-20:
ahh .. I think we're getting somewhere .. it looks like the ++ statement we added had caused PortD pins to work this morning and now the Port C pins are working and the Port D pins have stopped again.

Here's what I see on the Serial Monitor output of all  PortC Pins - (this one is for Pin23) when pushing the button connected to Pin23

arduinoPin=23,  portBitNumber=7
---------------------------------------
Pin was interrupted: 0 times so far.
---------------------------------------
Pin was interrupted: 0 times so far.
---------------------------------------
Pin was interrupted: 0 times so far.
---------------------------------------
Pin was interrupted: 0 times so far.
---------------------------------------
Pin was interrupted: 0 times so far.
---------------------------------------
interruptCount=1
interruptCount=2
Pin was interrupted: 2 times so far.
---------------------------------------
interruptCount=3
interruptCount=4
Pin was interrupted: 4 times so far.
----------:

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

And here's what I see on the PortD Pins now after removing the ++ statement and pushing the button connected to Port 12:

arduinoPin=12,  portBitNumber=4
---------------------------------------
Pin was interrupted: 0 times so far.
---------------------------------------
Pin was interrupted: 0 times so far.
---------------------------------------
Pin was interrupted: 0 times so far.
---------------------------------------
Pin was interrupted: 0 times so far.
---------------------------------------
Pin was interrupted: 0 times so far.
---------------------------------------
Pin was interrupted: 0 times so far.
---------------------------------------
Pin was interrupted: 0 times so far.
----------
>>>>>>>>>>>>>>>>>>>>>>>

Cat30Mk3, 2015-11-21:
Mike, I was mislead by the fact that the debug variable interruptCount is the same as the one in the Sample.ino program.  When I add a second variable to the ISR it doesn't increment - I don't think the ISR is being called in the cases where I thought the problem was solved.

Cat30Mk3, 2015-11-23:
Hi Mike,
 I've been digging through your code a bit and I think I found an error, but it doesn't resolve my problem. In the utilitiy/ei_portc_speed.h file. This is just an assumption based on observed inconsistency in code. I made my proposed changed but saw no effect.

In the section that begins with :
#if defined MIGHTY1284
the line:
  if (interruptMask & _BV(5)) INTERRUPT_FLAG_PIN22++;
should be:
  if (interruptMask & _BV(6)) INTERRUPT_FLAG_PIN22++;
and the line:
  if (interruptMask & _BV(5)) INTERRUPT_FLAG_PIN23++;
should be:
  if (interruptMask & _BV(7)) INTERRUPT_FLAG_PIN23++;

Also, here are the test detailed results from the last debug you requested - I'm reporting both C and D ports all pins still failed to interrupt based on what I mentioned in my previous comment. The variable interruptCount is increasing when PortC pins are changed (due to the debug code) but the ISR interruptFunction in Simple.ino is not being executed for these ports (based on adding second variable++ to interruptFunction).

I also swapped the Christensen Mighty1284 pins_arduino.h file  for the one provided with the LowPowerLabs MoteinoMega corefiles  just to confirm Felix's changes weren't the cause.  This did not resolve the issue either.

/* debugging test results 21/11/15 09:30
 *  using enableInterrupt(ARDUINOPIN | PINCHANGEINTERRUPT, interruptFunction, CHANGE);
 *  Debug code  "interruptCount++;" following "portSnapshotC = current;" in
 *  "EnableInterrupt.h"
ARDUINOPIN 7,  arduinoPin=7,   portBitNumber=7, PASS, NO interruptCount increments from .h
ARDUINOPIN 6,  arduinoPin=6,   portBitNumber=6, PASS, NO interruptCount increments from .h
ARDUINOPIN 5,  arduinoPin=5,   portBitNumber=5, PASS, NO interruptCount increments from .h
ARDUINOPIN 4,  arduinoPin=4,   portBitNumber=4, PASS, NO interruptCount increments from .h
ARDUINOPIN 3,  arduinoPin=3,   portBitNumber=3, PASS, NO interruptCount increments from .h
ARDUINOPIN 2,  arduinoPin=2,   portBitNumber=2, PASS, NO interruptCount increments from .h
ARDUINOPIN 1,  arduinoPin=1,   portBitNumber=1, PASS, NO interruptCount increments from .h
ARDUINOPIN 0,  arduinoPin=0,   portBitNumber=0, PASS, NO interruptCount increments from .h
ARDUINOPIN A0, arduinoPin=24,  portBitNumber=0, PASS, NO interruptCount increments from .h
ARDUINOPIN A1, arduinoPin=25,  portBitNumber=1, PASS, NO interruptCount increments from .h
ARDUINOPIN A2, arduinoPin=26,  portBitNumber=2, PASS, NO interruptCount increments from .h
ARDUINOPIN A3, arduinoPin=27,  portBitNumber=3, PASS, NO interruptCount increments from .h
ARDUINOPIN A4, arduinoPin=28,  portBitNumber=4, PASS, NO interruptCount increments from .h
ARDUINOPIN A5, arduinoPin=29,  portBitNumber=5, PASS, NO interruptCount increments from .h
ARDUINOPIN A6, arduinoPin=30,  portBitNumber=6, PASS, NO interruptCount increments from .h
ARDUINOPIN A7, arduinoPin=31,  portBitNumber=7, PASS, NO interruptCount increments from .h
ARDUINOPIN 10, arduinoPin=10,  portBitNumber=2, FAIL, NO interruptCount increments from .h
ARDUINOPIN 11, arduinoPin=11,  portBitNumber=3, FAIL, NO interruptCount increments from .h
ARDUINOPIN 12, arduinoPin=12,  portBitNumber=4, FAIL, NO interruptCount increments from .h
ARDUINOPIN 13, arduinoPin=13,  portBitNumber=5, FAIL, NO interruptCount increments from .h
ARDUINOPIN 14, arduinoPin=14,  portBitNumber=6, FAIL, NO interruptCount increments from .h
ARDUINOPIN 15, arduinoPin=15,  portBitNumber=7, FAIL, NO interruptCount increments from .h
ARDUINOPIN 16, arduinoPin=16,  portBitNumber=0, FAIL, interruptCount increments from .h
ARDUINOPIN 17, arduinoPin=17,  portBitNumber=1, FAIL, interruptCount increments from .h
ARDUINOPIN 18, arduinoPin=18,  portBitNumber=2, FAIL, interruptCount increments from .h
ARDUINOPIN 19, arduinoPin=19,  portBitNumber=3, FAIL, interruptCount increments from .h
ARDUINOPIN 20, arduinoPin=20,  portBitNumber=4, FAIL, interruptCount increments from .h
ARDUINOPIN 21, arduinoPin=21,  portBitNumber=5, FAIL, interruptCount increments from .h
ARDUINOPIN 22, arduinoPin=22,  portBitNumber=6, FAIL, interruptCount increments from .h
ARDUINOPIN 23, arduinoPin=23,  portBitNumber=7, FAIL, interruptCount increments from .h

I'm sure you're busy - just wanted to document my latest status.

John

GreyGnome, 2015-12-05:
Ok I'm back on this problem. I have fixed the bugs you found in utility/ei_portc_speed.h - Thanks. Now, on to the interrupting: My understanding at this point is that for Port C, the interruptFunction is not being called. But you know it's interrupting, because the debug variable I asked you to include is being incremented once. 

Furthermore, Port D is not interrupting at all.

Is this correct?

GreyGnome, 2015-12-05:
Right now I'm a little suspicious of some of the Arduino calls that I'm making in Simple.ino. I will come up with AVR-equivalents, which will involve toggling bits in registers. That will verify that we are actually setting the pins as expected.

GreyGnome, 2015-12-05:
I have a new Simple.ino sketch that I pushed to Github. Please download it and try it out. It has to be setting the pin modes correctly, so I'm sure that part at least will work. Just let me know if you see any different results.

Cat30Mk3, 2015-12-06:
thanks Mike - I've been away myself, but now back and available to test - I'll download the new code this evening and let you know - glad to hear you have a  1284p too - ti would be good to confirm you experience the same problem

Cat30Mk3, 2015-12-07:
Ok, I downloaded the full zip package and replaced the previous library with the new one tonight. Here's what I found:
(1) behaviour is the same as before using the new Simple.ino: Pin change interrupts work correctly for Ports A and B, but fail on Ports C and D. Specifically for Port C, Pins 10 and 11 (Int0 & Int1) work for the default external Interrupt but when pin change is forced they both fail. Pin 8 & 9 not tested.  Port D all pins fail to interrupt.
(2) I checked the file ei_portc_speed.h and found the error I previously identified  has not been corrected (but when I made the correction it did not solve the reported  problem for port C or port D.)
(3) I discovered another cut and paste type error in the file ei_Pinchange1284.h - I believe in lines 123 and 124  the variable should be risingPinsPORTD.... instead of risingPinsPORTB... again when I made the correction it did not solve the reported problem for port C and port D.)
Let me know if there's anything else I can try for you.
John

GreyGnome, 2015-12-09:
Ok, thanks. This means the problem is in the library. Ugh, I'll have a look.

SteveH01, 2016-04-25:
Hello.  I recently purchased a 1284P breakout for an LED project I'm working on.  I'm planning to use the EnableInterrupt library with a few pins on PORTC.  I'm using version 0.9.4 and I'm getting a Software Serial conflict with vector_4 whenever I include the library.  I'm using a version that I downloaded in December.  Is there a newer release than 0.9.4 that I need to install?  I'm happy to help troubleshoot the code if you need another perspective.

Thanks,
Steve Hurd
Floyds Knobs, IN

Cat30Mk3, 2016-04-26:
I’m running 0.9.4 . ‘just checked the Github site – it looks like Mike has released an untested version to address some issues related to 1284p processors.

 

I’ll give it a try as soon as  have a chance and report back.

 

John Witherspoob

SteveH01, 2016-04-27:
Sounds good!  As I said, if you need another tester I'm happy to help!

Steve Hurd

Cat30Mk3, 2016-04-27:
I downloaded and installed  a fresh zip tonight and ran through a complete test - I found no change from previous results.

/*test results 27/04/16 00:40
* Simple.ino MoteinoMega  pushbutton 'pin to gnd' 
Pass Pins: 7,6,5,4,3,2,1,0,A0,A1,A2,A3,A4,A5,A6,10,11,
Fail Pins:12,13,14,15,16,17,18,19,20,21,22,23
Not Tested Pins: 8,9 (in use for Serial Monitor communications)
*/

John

Cat30Mk3, 2016-04-27:
I’m not sure how the Github notifications work, but to confirm, I completed an all pin test with the current .zip package last night.

Port C and D still are still not interrupting with the exception of pins 10 and 11.

 

John

GreyGnome, 2016-04-30:
Ok, I found a serious bug in the PortC and D Pin Change Interrupt ISR code for the 1284 chip. Please download the EnableInterrupt.h file from the github repo and give it a try. The reason ports 10 and 11 were working is because those are External Interrupts, and the ISRs are completely different. Egg on my face!

GreyGnome, 2016-04-30:
Sorry it has taken me so long. I got laid off, there were the holidays, I went to Mexico, I started another project, and I got a new job. That's my meta-excuse. I believe I found the culprit, though. That's the good news.

Cat30Mk3, 2016-04-30:
Hey that's great Mike .. well, at least the new job and finding the bug! 

Sailboat launch tomorrow so testing will have to wait until later this weekend, but I'll get to it as soon as possible.  I've gotten a  lot of mileage out of the current version, but it will be great to have access to the full compliment of pins.

John Witherspoon

SteveH01, 2016-04-30:
Thanks Mike!  I'll try to test it this weekend as well.

Steve Hurd

Cat30Mk3, 2016-04-30:
SUCCESS!!  I confirm that all pins on Ports A, B, C and D   - can successfully raise an interrupt on state change. (I didn't test Ser0 - pins 8 & 9)

I'm satisfied this issue has been resolved.

'way to go Mike!

closed 2016-04-30 by Cat30Mk3

Cat30Mk3, 2016-04-30:
Hi Steve. After completing my tests and confirming all ports are working now I closed the issue in Githib. I should have waited for your results, please confirm back when you have had a chance to test. I can reopen the issue if you find problems.

John

John Witherspoon
(Res) (905) 420-6399
(Cell) (416) 528-1761
john.witherspoon@sympatico.ca

SteveH01, 2016-04-30:
Yay!

SteveH01, 2016-05-01:
Will do.  I'll try it out today.

SteveH01, 2016-05-01:
After looking at my layout, I saw that I had ran my rotary encoder
connections to the analog pins.  I have 3 available pins on port D so I
just jumpered the 3 analog pins to the digital pins.  As long a I don't use
the analog pins for anything I should still be able to trigger a Port D PC
interrupt with them connected this way, correct?

Steve Hurd

Cat30Mk3, 2016-05-01:
Ok, but why not just raise interrupts on the analog pins? 
Port A works fine with EnableInterrupt.

Either way, watch out for bounces with the rotary encoder.

John Witherspoon
(Res) (905) 420-6399
(Cell) (416) 528-1761
john.witherspoon@sympatico.ca

SteveH01, 2016-05-02:
Your FAQ said the analog ports don't work with PC interrupts.  Did I read
that wrong?

Cat30Mk3, 2016-05-02:
Hey Steve.. I re-read the FAQ that you mentioned.  It doesn't actually say that analog pins are not supported. It says that analog signals are not supported. I think it means that it will not trigger based on voltage level of input, but will trigger on high to low or low to high transitions. 

Also I checked Mike's very detailed and updated wiki where it clearly indicated the analog pins are supported (for digital interrupts) for the 1284p and many other chip types.

So you should be able to run your rotary encoder connected to the analog pins.

Hope this helps .. Let me know if you are successful.

John Witherspoon
john.witherspoon@sympatico.ca

SteveH01, 2016-05-02:
Thanks, John!  Must have missed that.  I'll try to test it when I have a
free evening this week.

Steve Hurd

Cat30Mk3, 2016-05-02:
Steve ... If you provide your email address we can chat without channeling our conversation  through Github. Mine is below. 

John
john.witherspoon@sympatico.ca

GreyGnome, 2016-05-03:
"Analog pins" are just another feature available on those pins. Just as you can use a pin for input OR output, some have UARTs attached to them, some are used for I2C, some for SPI, etc.... the beauty of the state of the art in MCUs is that not only are they cheap, they're incredibly adaptable. Don't let the fact that they're called "analog pins" confuse you... if you use them in their default digital mode, they act just like every other digital input on the ATmega, because they are digital inputs.

GreyGnome, 2016-05-03:
Glad to see everyone is please, and forgives the age and stinkiness of this bug. :-)

SteveH01, 2016-05-03:
Thanks Mike!  I'm coming at this from an IT background and still learning
about the hardware side of things.

Steve Hurd

GreyGnome, 2016-05-03:
Me too! IT is my middle name, Love the low level bits and bytes, though, as
you can tell.