GreyGnome/EnableInterrupt #32: Sleep mode issues

damonvp, 2016-05-02:
I realize this has been reported prior, however Im not getting how to resolve it.
I have a Rotary Encoder/Switch
I want the switch to enter and exit sleep mode.
I have the switch on pin4 of my uno.

void setup()
{
...
	enableInterrupt(2, intrEncChange1, CHANGE); //encoder interrupts
	enableInterrupt(3, intrEncChange2, CHANGE);
	enableInterrupt(4, intrEncChangeS, CHANGE);
...
}

void intrEncChangeS()
{
	if (sw)
		delay(1);
	if (digitalRead(SencPin) == encSW)
		return;
	encSW = !encSW;
	if (encSW)
		sleepNow();
	sw = false;
}
void intrEncChange1() //Read on interrupt Right turn - Fast +5
{
	if (moving)
		delay(1);
	if (digitalRead(encPin1) == enc1)
		return;
	enc1 = !enc1;
	if (enc1 && !enc2)
		encVal += 5;
	moving = false;
	update();
}
void intrEncChange2() //Read on interrupt Left turn - Slow -1
{
	if (moving)
		delay(1);
	if (digitalRead(encPin2) == enc2)
		return;
	enc2 = !enc2;
	if (enc2 && !enc1)
		encVal -= 1;
	moving = false;
	update();
}
void sleepNow()
{
        disableInterrupt(4); 
	enableInterrupt(4, wakeUpNow, LOW);
	set_sleep_mode(SLEEP_MODE_STANDBY);// Choose our preferred sleep mode
	sleep_enable();// Set sleep enable (SE) bit
        sleep_mode();// Put the device to sleep
	sleep_disable();// Upon waking up, sketch continues from this point.
	disableInterrupt(4); 
        enableInterrupt(4, intrEncChangeS, CHANGE);
}

How do I code this so that the switch works the way I need it to in order to resume from sleep mode?

markappa, 2016-05-06:
You should not put the device to sleep from an interrupt routine.
Try just setting some boolean flag in the ISR and then managing the sleep mode in the main loop.

GreyGnome, 2016-05-07:
Thanks, Mark. Also note damonvp that you cannot use delay() in an interrupt because it depends on a timer which raises interrupts, and inside the interrupt, interrupts are off.

GreyGnome, 2016-06-16:
Closing due to it looks closeable. :-)

closed 2016-06-16 by GreyGnome