GreyGnome/EnableInterrupt #33: multiple definition of `__vector_1' when compiling with MySensors

markappa, 2016-05-06:
I am getting this errors:
```
C:\Users\marco\AppData\Local\Temp\buildca530f920e52af2e7c8f67a769a556af.tmp/core\core.a(WInterrupts.c.o): In function `__vector_1':
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/WInterrupts.c:309: multiple definition of `__vector_1'
sketch\MySensor_Keypad.ino.cpp.o:C:\Users\marco\Documents\MySensors\libraries\EnableInterrupt/EnableInterrupt.h:1284: first defined here

C:\Users\marco\AppData\Local\Temp\buildca530f920e52af2e7c8f67a769a556af.tmp/core\core.a(WInterrupts.c.o): In function `nothing':
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/WInterrupts.c:35: multiple definition of `__vector_2'
sketch\MySensor_Keypad.ino.cpp.o:C:\Users\marco\Documents\MySensors\libraries\EnableInterrupt/EnableInterrupt.h:1011: first defined here
```

I did not figure out easily how __vector_1 and __vector_2 are defined/used since there are preprocessor macros involved.

- The EnableInterrupt examples compile correctly in my environment. 
- I am compiling for board Arduino Pro or Pro Mini (ATmega328 3.3V 8MHz).
- I am using the [MySensors ](http://www.mysensors.org/)library. **It seems to be conflicting with this.**

Any idea on how to solve?
Thank you!

markappa, 2016-05-06:
Ok, I saw some discussion on this issue [here](https://forum.mysensors.org/topic/1409/multiple-interrupts/9). 
Since I am not using internal interrupts, I resolved adding these lines in my sketch:

```
#define EI_NOTINT0
#define EI_NOTINT1
```

closed 2016-05-06 by markappa

GreyGnome, 2016-05-07:
Well that was quick- Nice work! Glad to see the library was able to handle the situation.