GreyGnome/EnableInterrupt #35: Compiling error conflict with RFM69.h

sid1202, 2016-10-11:
When I try to include EnableInterrupt with HFM69.h I get multiple definition of:
__vector_1
__vector_2

Are these ISR that I should modify in the RFM69 library ?

Sid

GreyGnome, 2016-10-14:
I'm not sure at the moment... I am occupied for a couple of days.

But yes, there is a conflict with the ISR's. If you look at my website I
tell you how to disable the definitions in EnableInterrupt. As I don't know
the HFM69.h library, I can't tell you what can/should change, but I do know
that something needs to.

Check my wiki for information about ISR conflicts. Let me know if you need
some help (but as I said it will be a day or two).

sid1202, 2016-10-15:
Thank you for your prompt reply.

I will look into it. There is no rush as I am tackling a few projects at the same time.

If I can't figure it out then I will request more help.

Thank you very much

Sidney

GreyGnome, 2017-09-18:
Closed as this has been months without updates.

closed 2017-09-18 by GreyGnome