GreyGnome/EnableInterrupt #37: Fake Pin of atmega2560 conflicts with pins from datasheet

MulajEgzon, 2016-11-17:
pin: fake70 PJ2 this is Pin Change Interrupt PCINT11
pin: fake71 PJ3 this is Pin Change Interrupt PCINT12
pin: fake72 PJ4 this is Pin Change Interrupt PCINT13
pin: fake73 PJ5 this is Pin Change Interrupt PCINT14
pin: fake74 PJ6 this is Pin Change Interrupt PCINT15
pin: fake75 PE6 this is External Interrupt INT6
pin: fake76 PE7 this is External Interrupt INT7
--------------------------------------------------------
Must BE ||
pin: fake65 PJ2 this is Pin Change Interrupt PCINT11
pin: fake66 PJ3 this is Pin Change Interrupt PCINT12
pin: fake67 PJ4 this is Pin Change Interrupt PCINT13
pin: fake68 PJ5 this is Pin Change Interrupt PCINT14
pin: fake69 PJ6 this is Pin Change Interrupt PCINT15
pin: fake8 PE6 this is External Interrupt INT6
pin: fake76 PE7 this is External Interrupt INT7

http://www.atmel.com/Images/Atmel-2549-8-bit-AVR-Microcontroller-ATmega640-1280-1281-2560-2561_datasheet.pdf
https://www.arduino.cc/en/uploads/Hacking/PinMap2560sma_.png

GreyGnome, 2016-12-01:
> Microcontroller-ATmega640-1280-1281-2560-2561_datasheet.pdf

You have given me the real ATmega2560 pin numbers that correspond with
those interrupts. But they conflict with the Arduino pin numbers. For
example, Arduino digital pin 69 is the same as Arduino Analog pin 15 which
is the ATmega2560 pin 82.

The pins I give are "fake" in the Arduino world, not in the ATmega2560
pinout.

If I use your numbers, however, then (for example) "fake" pin 8, aka PE6,
would conflict with the real Arduino pin 8... and in any event, it would
break the array digital_pin_to_port_bit_number_PGM.

Also, the definitions eg
#define ARDUINO_PIN_J2 70
would break as well.

But maybe I'm wrong. Do you have a bug that demonstrates bad behavior?

MulajEgzon, 2016-12-02:
Okay now i understand,  you did this just for ArduinoMega not for ATmega2560 chip.
No, I do not have any bug ! 
I'm using this library and it is very helping but I want to put all those pins in work and I was confused and did not know how to make them Work on ATmega2560 !
Could you please give me a hit how to put all PCInt pins at work as interrupts in ATmega2560 ?

Thank you very much for your Work (y).

GreyGnome, 2017-09-06:
This is way late, but my hint is as follows: Find a diagram of an ArduinoMega, and choose the pins that you're concerned about. You'll have to translate them into the ATmega2560 in your own mind. Then, once you've decided you've found the proper pin, you can used the library as you wish.

closed 2017-09-06 by GreyGnome

MulajEgzon, 2017-09-14:
Thank you. Now all pins are working!