GreyGnome/EnableInterrupt #39: Useless.h #include <EnableInterrupt.h> bug

brandonwkipp, 2017-01-04:
Hi, I'm using EnableInterrupt as a submodule for one of my projects. I found a compiler bug regarding EnableInterrupt/examples/SimpleWithLibrary/Useless.h; the verbose output is as follows:

In file included from /var/folders/70/7_syjtnx1zb_t62jc3qwvk_w0000gn/T/buildb1edcb75b05118bd4250494061fe263d.tmp/sketch/EnableInterrupt/examples/SimpleWithLibrary/Useless.cpp:3:0:
/var/folders/70/7_syjtnx1zb_t62jc3qwvk_w0000gn/T/buildb1edcb75b05118bd4250494061fe263d.tmp/sketch/EnableInterrupt/examples/SimpleWithLibrary/Useless.h:15:29: fatal error: EnableInterrupt.h: No such file or directory
 #include "EnableInterrupt.h"

And after a few minutes tinkering, I found a solution. Changing line 15 from **#include <EnableInterrupt.h>** to **#include "../../EnableInterrupt.h"**. Not really sure why it gave me errors considering all the other .ino files in the examples directory seem to find the file just fine.

GreyGnome, 2017-09-18:
I am unable to reproduce this error. I tried it with both the Arduino 1.8.4 IDE, and using Arduino-Makefile. 

When you compile in the IDE, make sure that "Show verbose output during compilation" is turned on, and run a compile. Grab all the output from the compile and add it to a Gist (https://gist.github.com/). Then link to it here. If the EnableInterrupt library is properly installed, you should see it included in the compile command of g++, as one of the "-I" options.

brandonwkipp, 2017-09-30:
I downloaded the newest version of the IDE and ran through a verify/compile and the issue is gone. Not sure what the problem was, but you can close this issue. Thanks!

closed 2017-09-30 by brandonwkipp