GreyGnome/EnableInterrupt #45: WInterrupts.c.o (symbol from plugin): In function `attachInterrupt': (.text+0x0): multiple definition of `__vector_1'

DaMicrok, 2017-10-19:
Massive apologies in advance, first time I've posted a problem. Like the title says I get a compiling error due to attachInterrupt and multiple definition of vector 1 and 2. I've tried using different digital pins and including #define EI_NOTINT0 and #define EI_NOTINT1 in my sketch but I do not know what is going wrong. I've read that you are not meant to use serial.print but that surely can't be the problem? because serial print is a major function of Arduino prototyping.

I am using 3 pins to read 3 Hall effect sensors and migrated from using attachInterrupt to enableInterrupt as the prior does not accommodate more than 2 pins.

Below is the error message

-------------------------------------------------------------------------------------------------------------------------------------------
Arduino: 1.8.3 (Linux), Board: "Arduino/Genuino Uno"

In file included from /local/XXXX/Arduino/flow_oct_v1/flow_oct_v1.ino:1:0:
/local/XXXX/Arduino/libraries/EnableInterrupt-master/EnableInterrupt.h:22:125: note: #pragma message: NOTICE: *** EnableInterrupt library version pre-0.9.6. This is not a problem. Keep calm, and carry on. ***
 #pragma message("NOTICE: *** EnableInterrupt library version pre-0.9.6. This is not a problem. Keep calm, and carry on. ***")
                                                                                                                             ^
WInterrupts.c.o (symbol from plugin): In function `attachInterrupt':
(.text+0x0): multiple definition of `__vector_1'
sketch/flow_oct_v1.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
WInterrupts.c.o (symbol from plugin): In function `attachInterrupt':
(.text+0x0): multiple definition of `__vector_2'
sketch/flow_oct_v1.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Uno.

-------------------------------------------------------------------------------------------------------------------------------------------And this is my sketch

#include <EnableInterrupt.h>

#define EI_NOTINT0
#define EI_NOTINT1

/*
Liquid flow rate sensor -DIYhacking.com Arvind Sanjeev

Measure the liquid/water flow rate using this code. 
Connect Vcc and Gnd of sensor to arduino, and the 
signal line to arduino digital pin 2.
 
 */

byte statusLed    = 13;

// byte sensorInterrupt = 0;  // 0 = digital pin 2
byte sensorPin1       = 5;

// The hall-effect flow sensor outputs approximately 4.5 pulses per second per
// litre/minute of flow.
float calibrationFactor = 4.5;

volatile byte pulseCount1;  

float flowRate1;
unsigned int flowMilliLitres1;
unsigned long totalMilliLitres1;

unsigned long oldTime;

byte sensorPin2       = 4;
// The hall-effect flow sensor outputs approximately 4.5 pulses per second per
// litre/minute of flow.

volatile byte pulseCount2;  

float flowRate2;
unsigned int flowMilliLitres2;
unsigned long totalMilliLitres2;

byte sensorPin3       = 7;

volatile byte pulseCount3;  

[... 169 lines cut ...]
  }
}

/*
Insterrupt Service Routine
 */
void pulseCounter1()
{
  // Increment the pulse counter
  pulseCount1++;
}
void pulseCounter2()
{
  // Increment the pulse counter
  pulseCount2++;
}
void pulseCounter3()
{
  // Increment the pulse counter
  pulseCount3++;
}

Apologies again as I can only presume there must be a painfully obvious solution to those that know.
And thanks.
Michael

GreyGnome, 2017-10-21:
Hello, no problem... we've all been there. You have a line `detachInterrupt(7);` which needs to be changed to `disableInterrupt(7);`. That will fix your sketch.

You can use Serial.print()... as a matter of fact, my examples do so. You just shouldn't put it inside an Interrupt Service Routine. Your technique of updating the pulse counters is just fine, as is your using Serial.print() to display output.

closed 2017-10-21 by GreyGnome

DaMicrok, 2017-10-23:
Oh my, haha, what a schoolboy error!
Thank you GreyGnome, I will test this out asap 👍

DaMicrok, 2017-10-23:
Yep, tried and tested. Works like a dream.

I am amazed and full of gratitude that you took time out to respond to such a newbie mistake and you did so very quickly. Thank you so so much. :1st_place_medal:

dukepow, 2021-03-24:
Using radiohead rh_ask.h the exact same thing happens. I don't know of this error.

GreyGnome, 2021-03-27:
This must be because you are not excluding the interrupts properly in
enable interrupt. Have you, double checked the documentation on how to
exclude interrupts?