GreyGnome/EnableInterrupt #46: Doesn't work together with SoftwareSerial Library

donyayo, 2017-11-06:
This Library is having conflict with the software serial library. Is there a way around using the library with the softwareserial library?

SRGDamia1, 2017-11-07:
Without modifications, this library will not work with SoftwareSerial:  https://github.com/GreyGnome/EnableInterrupt/wiki/FAQ#no-really-the-enableinterrupt-library-does-not-work-with-the-software-serial-library

Like the wiki suggests, [AltSoftSerial ](https://github.com/PaulStoffregen/AltSoftSerial), is a better replacement for SoftwareSerial.  It has a limited selection of boards it supports, but I think it almost completely overlaps with the boards supported by EnableInterrupt.  Each board only has one or two pins that AltSoftSerial can use for Tx/Rx.  But, unless you for some reason can't use those pins or must use the timer for something else, AltSoftSerial really is more stable, accurate, and has far fewer conflicts.

[NeoSWSerial](https://github.com/SlashDevin/NeoSWSerial) is another SoftwareSerial replacement that by default controls all of the interrupts, but is built to allow external interrupt control by uncommenting line 108 in NeoSWSerial.h.  NeoSWSerial only works at 9600 (default), 19200 and 38400 baud.  Slower is better for NeoSWSerial.

If you're really set on using SoftwareSerial, you have modify SoftwareSerial.cpp (lines 214-245ish) so it can only control the interrupt vector for the pins you want to use for software serial.  Then you have to use the commands from the "[Save Memory](https://github.com/GreyGnome/EnableInterrupt/wiki/SaveMemory)" instructions to disable the EnableInterrupt control of those same vectors.

I tried making a [modified version of SoftwareSerial ](https://github.com/EnviroDIY/SoftwareSerial_ExternalInts) that hands off all interrupt control to another library like EnableInterrupt, but it really didn't work.  With all of the extra handing off, the timing is too far off for even marginal reliability.

thehapyone, 2017-11-07:
You are right, the AltSoftSerial Library is a way better replacement for the SoftwareSerial one. I might consider using it or just write the Interrupt code myself. 
Thank you.

closed 2017-11-07 by thehapyone