GreyGnome/EnableInterrupt #47: Wake from Sleep Atmega2560 INT7 (PE7) - Gboard Pro

Travis92, 2018-04-05:
Hi, first of all i wanted to thank you for this wonderful job of yours.
I am using a Gboard Pro by Itead which is basically an Atmega2560 with Nrf24 IRQ pin attached to pin PE7 of the atmega. I can use your example code for generating the ISR on that pin but if i put the MCU i can't wake it up by that interrupt. Could you please explain how to achieve so? I suppose it is related to some flag that i read about in the datasheet but i am not that into technicalities. 
Thanks in advance.

stickbreaker, 2018-04-05:
@travis92
> I suppose it is related to some flag that i read about in the datasheet but i am not that into technicalities.
Thanks in advance.

You must be a millennial 'do it for me, Its out of my safe zone.' You have the right idea where to look for your answer but are unwilling to expend 'your' time. You expect others to do it for you. 
I hope no-one responds with the answer.

Chuck.

Travis92, 2018-04-06:
Some people love to reinvent the wheel. Some others just ask if anyone have already done it/suggestions for doing it. If you don't want to answer just don't do it.

GreyGnome, 2018-04-07:
Sorry no I don't know how to do that. Perhaps some others would... the
interrupt library should be able to set up the interrupt, but the part
about waking from sleep is something I haven't done.

Travis92, 2018-04-17:
`
#include <SPI.h>
#include <digitalWriteFast.h>
#include "iBoardRF24.h"
#include "printf.h"
#include <avr/sleep.h>
#include <EnableInterrupt.h>

//
// Hardware configuration
//

// Modify this at your leisure. Refer to https://github.com/GreyGnome/EnableInterrupt/wiki/Usage#Summary
#define ARDUINOPIN 76

// Set up nRF24L01 radio on iBoard
iBoardRF24 myRadio(12, 11, 8, 7, 9, PE7);

uint64_t addresses[1] = {0x65646f4e31}; // Create address for 1 pipe.

byte radioData;

byte ACK;

byte interruptcount=0;

void setup() {
  Serial.begin(115200);
  Serial.println(F("RF24/Simple Receive data Test"));

  printf_begin(); //inizializza printf per il debug del nrf24l01
  myRadio.begin();  //inizializza nrf24l01

  myRadio.enableAckPayload(); //SetACK payload
  myRadio.setAutoAck(true);  //set autoACK

  myRadio.setChannel(108);  //Set channel
  myRadio.setPALevel(RF24_PA_MAX);  // Set the PA Level
  myRadio.setDataRate(RF24_2MBPS); //set data rate

  //myRadio.maskIRQ(1,1,0);               //mask all IRQ triggers except for receive (1 is mask, 0 is no mask)

  myRadio.openReadingPipe(1, addresses[0]); // Use the first entry in array 'addresses' (Only 1 right now)
  myRadio.startListening();

  myRadio.printDetails();
  enableInterrupt(ARDUINOPIN, svegliatiora, LOW); 

}

void loop() {
/* 
  while (myRadio.available())  // While there is data ready
  {

    myRadio.read( &radioData, sizeof(radioData) ); // read the incoming data
    // DO something with the data, like print it
    ACK = radioData + 100;
    Serial.print(F("OK, dati ricevuti: "));
    Serial.print(radioData);
    Serial.print("    Writing ACK: ");
    myRadio.writeAckPayload( 1, &ACK, sizeof(ACK) );
    Serial.println(ACK);
  }
*/
  Serial.println("Sveglio");
  Serial.println(interruptcount);
  delay(700);

  dormiora();
}

void dormiora()
{
  set_sleep_mode(SLEEP_MODE_PWR_DOWN); // tipo di "addormentamento" desiderato
  // (SLEEP_MODE_PWR_DOWN e' il piu' profondo, quello che provoca il maggior risparmio di energia) //
  // Serial.println ("mi addormento");
  
  Serial.println ("dormo");
  delay (500);
  sleep_enable(); // abilita l'attivazione della modalita' sleep
  // Enable sleep bit in the mcucr register
  // utilizzato nei cicli di for - used in "for" cycle
  //Arduino: sleep, wake ed interrupts //Arduino, sleep, wake, interrupts
  // L'interrupt di tipo 0 e' l'interrupt generato da uno stato LOW sulla porta 2. Nel momento in // cui viene ricevuto un interrupt viene automaticamente lanciata la routine "svegliatiora")
  // Use interrupt #0 (pin 2) and run "svegliatiora" function on interrupt
  //
  sleep_mode(); //attivazione della funzione sleep. Da ora il sistema e' in sleep e puo' esere // svegliato solo da un interrupt sulla porta 2- Here the device is actually put to sleep!!
  //
  // ******************************************************************************************* // questo e' il punto in cui il sistema riprende a lavorare subito dopo il risveglio e subito // dopo aver eseguito la routine "svegliatiora"
  // system start work here, after wake and "svegliatiora" routine //******************************************************************************************** //
  //disableInterrupt(ARDUINOPIN);

}

void svegliatiora()
{
//sleep_disable(); // disabilita la funzione sleep
interruptcount++;
// le due precedenti istruzioni devono essere le prime ad essere eseguite al momento del
// risveglio, in modo da evitare eventuali ulteriori interrupt derivanti dalla pressione
// prolungata del pulsante - These above two instruction must be execute at first after wake
}
//
//`

This is what I achieved so far. The MCU sleeps and wakes up when the packet is received on the IRQ pin (and triggers the interrupt function). It reboots. I don't know why so far but I'm going on.
If anyone wants to try any help is appreciated.

cheers