GreyGnome/EnableInterrupt #48: ATmega328PB

hgruenebe, 2018-04-06:
Hi GreyGnome, is it possible to get the Lib work with an 328PB ?
Would be great Thx.

GreyGnome, 2018-04-07:
Hmmm... Isn't this roughly equivalent to an ATmega328? Could you tell me
the differences in the Interrupt structure? I imagine they would be
equivalent.

The 328PB would have some additional QTouch features. I would say, if you
treat the 328PB like a 328 then you should be good.

hgruenebe, 2018-04-07:
Thank you for your response.
Good news.
It's working fine with my Pololu A-Star 328pb Board :-)
The main difference and reason for me to use the 328pb, is the second hw serial port.
I have to use your lib as the base of another lib, but I need for my projects two serial lines (one for meassuring, the scond for serial monitoring/ IDE transfer). 
Because EnableInterrupt does not work together with softwareserial or other alternatives, the 328pb would be an valid possibility to have an second hw serial port.
Yesterday I had some problems with the IDE an the right arduino board description. 
Today I gave him another try and it works.
I assume the most reasonable problem was the hw wiring of my breadboard :-((

GreyGnome, 2018-06-21:
I wonder if you have tried the AltSoftSerial library? See https://github.com/PaulStoffregen/AltSoftSerial. That said, there's nothing more to be done here so I will close this issue.

closed 2018-06-21 by GreyGnome