GreyGnome/EnableInterrupt #52: Teensy 3.2 compatibility?

amerkay, 2018-10-06:
Hi,

Quick question, how can I get it working with [Teensy 3.2](https://www.pjrc.com/store/teensy32.html)? 

I understandably get the `Unsupported Arduino platform` error.

Thanks!

GreyGnome, 2018-10-07:
I don't think it's necessary, because the Teensy 3.2 is ARM-based. The
library does support the Due, but that's only faked with a macro. I think
doing something for the Teensy would be equally as simple. Maybe you could
take a look at how the Due is handled.

The main reason for the library is because of the way the ATmega series
does its interrupts... it's very quirky. The library makes the interrupts
look "normal". You don't need to do that on an ARM based processor. On the
other hand, if you want software that that's compatible with both platforms
then I can see how you would need this feature.

Is that your use case?

GreyGnome, 2019-09-05:
Closed as inactive, and likely not an issue.

closed 2019-09-05 by GreyGnome