GreyGnome/EnableInterrupt #53: Compile error with external interrupts

jimbowley, 2018-10-19:
Thanks for this library, 

Should I be able to use both this and external interrupts together?

just including this library with my code causes compile error. 
I'm using a ProMicro (Leonardo) and  IDE 1.8.1

> 
#include <Keyboard.h>
#include <HID.h>
#include <Mouse.h>
//#include <EnableInterrupt.h>

#define pinA1 1   //Roller A
#define pinZ1 8   //Spinner

long cntX=0; //counter for mouse X-axis
long lastcntX=0;
float realX=0;
float incX= 0.5;  // This gives approx correct speed on missile command default sensitivity 20
float incZ= 0.1;  // high resolution spinner  needs low value here.

void setup() {
   
 // pinMode(pinZ1, INPUT_PULLUP);
 // enableInterrupt(pinZ1,doZCount, CHANGE);
  
  pinMode(pinA1, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(pinA1),doACountX,CHANGE);

  Mouse.begin();
  Keyboard.begin();
}

void loop() {
  //send mouse update to usb
  cntX = realX;
  if(cntX!=lastcntX ){
    Mouse.move((cntX-lastcntX),0);
    lastcntX=cntX;
  }
}

void doZCount() {  // spinner is moving
 realX = realX +incZ;
}

void doACountX() {  // ball is moving
  realX = realX +incX;
}

Mihai-B, 2018-12-19:
I am not able to run this with external interrupts as well. With the deprecated library I was able to do so. I am now wondering if this is a bug?

happysupply, 2018-12-20:
Same problem here. Tried compiling for Uno with both external and PinChange interrupts, through functions attachInterrupt() and enableInterrupt(). This didn't work.

When I replaced the attachInterrupts for enableInterrupts, the code did compile.

happysupply, 2018-12-24:
Hi Mike, many thanks for your quick reply. When I replaced attachInterrupt() with enableInterrupt(), I was worried that the interrupt trigger on (UNO) pins 2 and 3 were slowed down (because enableInterrupt is slower). Your answer clarifies I shouldn't worry about this.

Problem is solved now, as far as I'm concerned!

happysupply, 2018-12-24:
Exactly, that was my concern. Good to hear I shouldn't be concerned indeed. Many thanks again and happy holidays:)

GreyGnome, 2019-09-03:
Closing this as not an issue. :-)

closed 2019-09-03 by GreyGnome