GreyGnome/EnableInterrupt #55: Analog PIN - interrupt fired with minimal voltage change

ale-trevizoli, 2019-04-16:
Hi. I'm using Mega2560 and Analog pin A8 to A15.
I'm seeing strange behavior, using CHANGE, RISING or FALLING, any voltage change fire interrupt.
example, if it's on 0V and goes to 1V interrupt fire.
if it's 5V and goes to 4.9V interrupt fire.
Could you help me?

Sorry for my poor english!

Thanks a lot! This library save my life!

GreyGnome, 2019-04-17:
Hello,
I would not expect the interrupt to fire based on the levels you mentioned.
Are you sure there are no voltage spikes or dips? I think you need to scope
your pin, and watch it very closely. My guess is that it is dropping well
below 4.9V, if only for a microsecond, and it's rising above 1V- again, for
a very short time. But the Atmega will pick it up.

Also, do you have a resistor on the pin pulling it high (or low)? If you
have a switch or other high-impedance device connected to the pin, it may
not be pulling the voltage the way you expect.

In short: it sounds to me like the pins are not seeing what you expect. I
suspect a circuit problem, so use an oscilloscope to observe your signals.

ale-trevizoli, 2019-04-17:
I will confirm it, we are not using resistor on pin. I will put it and test again.
I thought it was the voltage variation, because if I put my finger on the pin, it triggers the interruption infinitely, and I used AnalogRead on pin inside function, and it show me random 0.1 to 0.5v decrease.
But I will test with PULL-UP and PULL-DOWN resistor.

GreyGnome, 2019-04-18:
Yes, the impedance on the ATmega is high. So if you put your finger on the
pin, I would expect to see many interrupts. This is from your body's
electrical field.

You should use either a pull-up or pull-down resistor, not both. Depending
on your circuit. Usually a pull-up resistor is used.

GreyGnome, 2019-09-03:
This is not an issues with the library. I'm gonna close it.

closed 2019-09-03 by GreyGnome