GreyGnome/EnableInterrupt #57: FALLING function is never fired

g105b, 2019-08-23:
Here is my setup function:

```
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(9600);
  while(!Serial);
  Serial.setTimeout(1000);

  pinMode(PIN_COUNTER_1, INPUT_PULLUP);
  digitalWrite(LED_BUILTIN, HIGH);
  enableInterrupt(PIN_COUNTER_1, c1_fall, FALLING);
  enableInterrupt(PIN_COUNTER_1, c1_rise, RISING);
}
```

Within `c1_rise` function, the `rise_counter` variable is incremented successfully.

Within `c1_fall` function, the `fall_counter` variable is incremented, but it is never called (counter remains on 0).

jimbowley, 2019-08-23:
Can you have more than one interrupt type on a single pin (at the same time)?   I wouldn't have guessed that.

GreyGnome, 2019-08-23:
Bowler, Bowley is correct. 1 interrupt type per pin please. You want
CHANGE. You'll need to check the state of your pin, within the interrupt
handler, to know if you rose or fell getting there. As the docs say, "These
interrupts can be set to trigger on RISING, or FALLING, or both ("CHANGE")
signal levels, or on LOW level..."

"OR' only- no "AND"s. So it's, "RISING" *or* "FALLING" *or* "CHANGE".

Good luck. Let me know how it goes.

GreyGnome, 2019-08-23:
...*or* "LOW". I forgot the "LOW".

GreyGnome, 2019-08-23:
Ugh, "LOW" is only on External Interrupts. That should be noted.

g105b, 2019-08-26:
Thank you, that answered my issue and the project is working fine now.

Just to be pedantic,

> "These interrupts can be set to trigger on RISING, or FALLING, or both ("CHANGE").

The sentence does literally say that they can be set to trigger on rising, falling or both. I fully understand the issue I was having now, and have to say that your documentation is actually really refreshingly well written, but that one sentence is obviously the one that caught me out. I read it as saying I could trigger on both.

Thanks for the library, and thanks for your answers! Have a great week.

closed 2019-08-26 by g105b

GreyGnome, 2019-08-26:
OK, I'll spell it out just a little bit more in the docs. Thanks for the
feedback.

g105b, 2019-08-27:
May I suggest:

> These interrupts can be set to trigger on one of three values: RISING, FALLING or CHANGE (for both).

GreyGnome, 2019-09-03:
Nice, I like it!