GreyGnome/EnableInterrupt #60: How to make pins 20 and 21 of 328P work ?

kbssa, 2019-12-19:
Hi,

I need to include the pins 20 and 21 (Crystals pins) of atmega328P in your library.

I have tried by my own, but with no success.

Could you point me to right direction, so I can make it work ?

Thanks

GreyGnome, 2019-12-27:
It can probably be done, I'm not sure how easily though. I'll take a look...

kbssa, 2019-12-28:
Good to hear that, I only need this to finish my  project.

Thank you very much

GreyGnome, 2019-12-29:
In order for me to properly do this, I'll probably need an atmega328p
standalone. I've got some on order but they won't be here for 3 weeks. In
the meantime, I may try to get the attiny2414 to work with this library,
and there I'll be allowing the crystal oscillator pins to be used as GPIO.
It should be similar.

GreyGnome, 2019-12-29:
...Oops, I mean attiny84. Not attiny2414.

kbssa, 2020-01-02:
> In order for me to properly do this, I'll probably need an atmega328p standalone. I've got some on order but they won't be here for 3 weeks. In the meantime, I may try to get the attiny2414 to work [...]

Could you point me to what I need to change to make it work ? Or it's more than a few changes ?

Thanks

GreyGnome, 2020-01-08:
OK, I'm doing some looking around. First, is this code (line numbers included):
`1243     portMask=pgm_read_byte(&digital_pin_to_bit_mask_PGM[arduinoPin]);
1244     portNumber=pgm_read_byte(&digital_pin_to_port_PGM[arduinoPin]); `

So there are 2 problems: arduinoPin for PB6 and PB7 (ATMEGA328 pins 9 and 10) is not defined. On the Arduino, it only goes to pin 19. So you would need to have an "if" statement in the above. If the pin was 20 or 21, return the proper portMask and portNumber.
Then, I think here:
`1841 #if ! (defined ARDUINO_328) && ! (defined EI_ATTINY24) && ! (defined EI_ATTINY25)
1842   if (interruptMask & _BV(6)) { arduinoInterruptedPin=ARDUINO_PIN_B6; arduinoPinState=current & _BV(6); portBFunctions.pinSix(); }
1843   if (interruptMask & _BV(7)) { arduinoInterruptedPin=ARDUINO_PIN_B7; arduinoPinState=current & _BV(7); portBFunctions.pinSeven(); }`
Specifically, line 1841, here's no reason for the ARDUINO_328 in there, because we DO want _BV(6) and _BV(7) to be valid.
Same with line 1854; remove the `! (defined ARDUINO_328)`:
`1854 #if ! (defined ARDUINO_328) && ! (defined EI_ATTINY24) && ! (defined EI_ATTINY25)`
Between line 186 and 187, insert:
`#define ARDUINO_PIN_B6 20 // A lie! Because the Arduino has an xtal here, but we want to use
#define ARDUINO_PIN_B7 21 // these pins on non-Arduino Atmega328's`
Between lines 222 and 223:
`6,
7`
...and fix line 221 by appending a comma.

Between lines 237 and 238, insert:
`interruptFunctionType pinSix;
interruptFunctionType pinSeven;`

Change line 241 to include more NULL's:
`functionPointersPortB portBFunctions = { NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL };`

I think that's about it, but I'm not 100% sure without testing it. I have some 328p's on order.

kbssa, 2020-01-09:
Ok, Thank's a lot !

When I do some tests I post the results here.

GreyGnome, 2020-01-10:
No problem. Remember, for lines 1843 and 1844 you have to have an "if" statement and return what the port mask and bit number are verbatim... don't use the "digital_pin_to..." macros because they cannot handle the additional pins.