GreyGnome/EnableInterrupt #62: Conflict with SoftwareSerial.h library on UNO?

cap10ray, 2020-05-18:
Very nice piece of work, a big thanks. I seem to have a conflict when trying to use EnableInterrupt and SoftwareSerial libraries on an Arduino Uno.  I have an XBee3 module I'm using for communications (a longer distance that wifi), and need to monitor a waterflow sensor that sends pulsed output according to the flow.  It requires an interrupt to count pulses. If I disable SoftwareSerial I get a clean compile and the pulse counter works, but without the XBee support.  If I disable the EnableInterrupt, I clean compile with XBee communications, but no pulse counting.  
I don't know if you came across this, I seem to recall a comment in PInChangeInterrupt documentation relating to an issue with SoftSerial, but did not see that in your documentation.
Is there any workaround that you are aware of?
Thanks
Ray

DrIce2000, 2021-07-20:
Same issue with the Mega2560 and Softwareserial.

C:\Users\de530\AppData\Local\Temp\arduino_build_356207\libraries\SoftwareSerial\SoftwareSerial.cpp.o (symbol from plugin): In function `SoftwareSerial::read()':

(.text+0x0): multiple definition of `__vector_9'

C:\Users\de530\AppData\Local\Temp\arduino_build_356207\sketch\HiSpeed.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\de530\AppData\Local\Temp\arduino_build_356207\libraries\SoftwareSerial\SoftwareSerial.cpp.o (symbol from plugin): In function `SoftwareSerial::read()':

(.text+0x0): multiple definition of `__vector_10'

C:\Users\de530\AppData\Local\Temp\arduino_build_356207\sketch\HiSpeed.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\de530\AppData\Local\Temp\arduino_build_356207\libraries\SoftwareSerial\SoftwareSerial.cpp.o (symbol from plugin): In function `SoftwareSerial::read()':

(.text+0x0): multiple definition of `__vector_11'

C:\Users\de530\AppData\Local\Temp\arduino_build_356207\sketch\HiSpeed.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

Bibliothek EnableInterrupt in Version 1.1.0 im Ordner: C:\Users\de530\Documents\Arduino\libraries\EnableInterrupt  wird verwendet

Bibliothek SoftwareSerial in Version 1.0 im Ordner: C:\Users\de530\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\libraries\SoftwareSerial  wird verwendet

exit status 1

Fehler beim Kompilieren für das Board Arduino Mega or Mega 2560.

Thanks
Markus

GreyGnome, 2021-07-26:
Yes, this is a known issue. I discuss it here:
https://github.com/GreyGnome/EnableInterrupt/wiki/FAQ

Let me know if you have any questions after reading that.