GreyGnome/EnableInterrupt #64: Conflict with LMIC library

j45p41, 2021-07-21:
This is indeed a very useful library, infact the only library that I can find that allows me to use six interrupts to wake the Arduino. However when I use it with the LMIC library (Matthijs Kooijman ) the "Forward uplink data message" is not sent during initial registration, only if 
#include <EnableInterrupt.h> is commented out does it work. I am using a RFM95 with the following LMIC pin assignments:

const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 9,
    .dio = {3, 8, LMIC_UNUSED_PIN},
};

This must be something to do with the library sharing ISRs.  Could somebody point to which parts of the LMIC library so as not to have this conflict?

I have tried swapping 8 for A5 and it again only works when the line is commented. Any help would be greatly appreciated.

j45p41, 2021-07-21:
I have managed to get it to work by specifying
#define EI_NOTPORTD however it does not work with  #define EI_ARDUINO_INTERRUPTED_PIN which is needed for my project to function.

j45p41, 2021-07-22:
This is the best I can get to work with 6 available pin change interrupts - just enough! 

#define EI_ARDUINO_INTERRUPTED_PIN
#define EI_NOTPORTB  // DISABLE FOR LMIC
#define EI_NOTPORTD  // DISABLE FOR LMIC

GreyGnome, 2021-07-26:
Thanks for the update. Yeah, these little ATMega chips are strange with the
interrupts...!

j45p41, 2021-08-08:
Hi, 
so managed to get really far, until of course, I got to the point just before the end!

I have a situation where I have interrupts defined in the following way:

  enableInterrupt(CHANNEL3_IN_PIN, interruptFunction1, CHANGE);
  enableInterrupt(CHANNEL4_IN_PIN, interruptFunction, FALLING);
  enableInterrupt(CHANNEL5_IN_PIN, interruptFunction, FALLING);
  enableInterrupt(CHANNEL6_IN_PIN, interruptFunction, FALLING);
  enableInterrupt(CHANNEL7_IN_PIN, interruptFunction, FALLING);
  enableInterrupt(CHANNEL8_IN_PIN, interruptFunction1, CHANGE);

where